dronekit/dronekit-python #629: Can't seem to get SITL to takeoff with RC channel overrides

alexblack, 2016-05-25:
I'm trying to figure out how to fly the copter with RC channel overrides, it never leaves the ground, any ideas what I'm doing wrong? Thanks!

```
    lat = 49.2778638
    lon = -123.0633418

    sitl = SITL()
    sitl.download('copter', '3.3', verbose=True)
    sitl_args = ['-I0', '--model', 'quad', '--home={0},{1},584,353'.format(lat, lon)]
    sitl.launch(sitl_args, await_ready=True, restart=True)
    connection_string = 'tcp:127.0.0.1:5760'

    vehicle = dronekit.connect(connection_string, wait_ready=True, vehicle_class=DdcVehicle)

    # Don't let the user try to arm until autopilot is ready
    while not vehicle.is_armable:
      print(" Waiting for vehicle to initialise... (GPS={0}, Battery={1})".format(vehicle.gps_0, vehicle.battery))
      time.sleep(1)

    # Copter should arm in GUIDED mode
    vehicle.change_mode('STABILIZE')
    vehicle.armed2 = True

    while not vehicle.armed2:      
      print(" Waiting for arming...")
      time.sleep(1)

    #vehicle.simple_takeoff(1.5)

    vehicle.channel_override = { "3": 2000 }
    vehicle.flush()

    print " Ch3 override: %s" % vehicle.channel_override['3']

    while True:
      if vehicle.location.global_relative_frame.alt >= 1.4: 
        print('Reached target altitude: {0:.2f}m'.format(vehicle.location.global_relative_frame.alt))
        break
      else:
        print("Altitude: {0:.2f}m".format(vehicle.location.global_relative_frame.alt))
      time.sleep(0.5)
```

peterbarker, 2016-05-26:
On Wed, 25 May 2016, Alex Black wrote:

> I'm trying to figure out how to fly the copter with RC channel overrides, it never leaves the ground, any ideas what I'm doing wrong? Thanks!

Here's your example modified so that It Works For Me.  The ctitical 
problem was that you are doing something with "channel_override" rather 
han "channel.overrides"...

Standard disclaimers apply: using RC channels like this to move your 
aircraft around is probably a very bad idea....

#!/usr/bin/python

import dronekit
import dronekit_sitl
import time

lat = 49.2778638
lon = -123.0633418

sitl = dronekit_sitl.SITL()
sitl.download('copter', '3.3', verbose=True)
sitl_args = ['-I0', '--model', 'quad', 
'--home={0},{1},584,353'.format(lat, lon)]
sitl.launch(sitl_args, await_ready=True, restart=True)
connection_string = 'tcp:127.0.0.1:5760'

vehicle = dronekit.connect(connection_string, wait_ready=True)

# Don't let the user try to arm until autopilot is ready
while not vehicle.is_armable:
   print(" Waiting for vehicle to initialise... (GPS={0}, 
Battery={1})".format(vehicle.gps_0, vehicle.battery))
   time.sleep(1)

# Set vehicle mode
desired_mode = 'STABILIZE'
while vehicle.mode != desired_mode:
   vehicle.mode = dronekit.VehicleMode(desired_mode)
   time.sleep(0.5)

while not vehicle.armed:
     print("Arming motors")
     vehicle.armed = True
     time.sleep(0.5)

while True:
   vehicle.channels.overrides[3] = 2000
   print " Ch3 override: %s" % vehicle.channels.overrides[3]

   if vehicle.location.global_relative_frame.alt >= 140:
     print('Reached target altitude: 
{0:.2f}m'.format(vehicle.location.global_relative_frame.alt))
     break
   else:
     print("Altitude: 
{0:.2f}m".format(vehicle.location.global_relative_frame.alt))
   time.sleep(0.5)

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

alexblack, 2016-05-26:
Nice catch, thanks. Using the correct `channel_overrides` and calling it each time through the altitude check loop has it taking off.  Very helpful!

Biswajit1995, 2017-02-10:
Hey alexblack,

Im trying to make my drone fly without GPS and to move to certain distance by overriding RC commands, for maintaining position I am using optical flow camera and a lidar, but the thing is Im not able to arm my copter through my script.

I'm using this following code:

`#/usr/bin/env python
# -*- coding: utf-8 -*-
from dronekit import connect,VehicleMode,LocationGlobalRelative
from pymavlink import mavutil
import time
# Receiving from command line
import argparse
parser = argparse.ArgumentParser()
parser.add_argument('--connect',help="PORT_NO")
args = parser.parse_args()

# Connecting to the vehicle
connection_string = args.connect
print("Connecting to...% s" % connection_string)
vehicle = connect(connection_string, wait_ready=True)

#Function to arm and takeoff to a specified altitude
def arm_and_takeoff(aTargetAlt):
	print("Basic Prearm checks..dont touch!!")
	print("Waiting for vehicle to initialize")
	time.sleep(1)
	print("Arming Motors..")
	# Copter should arm in Guided-mode
	vehicle.mode = VehicleMode("STABILIZE")
	
	while not vehicle.is_armable:
		print("Waiting for vehicle to initialize...")
		time.sleep(2)

	vehicle.armed = True
	vehicle.flush()
        print("Is Armed:% s"% vehicle.armed)
	
	while not vehicle.armed:
		print("Waiting for arming...Mode:% s"% vehicle.mode)
		print("Is Armed:% s"% vehicle.armed)
		vehicle.armed = True
		time.sleep(2)
		
	print("Taking Off..")
	vehicle.simple_takeOff(aTargetAlt)

	while True:
		print("Altitude: ",vehicle.location.global_relative_frame.alt)
		#Break and return from function just below target altitude.
		if vehicle.location.global_relative_frame.alt>=aTargetAlt: 
			print("Reached Target Altitude..")
			break
		time.sleep(2)
	
vehicle.close()
arm_and_takeoff(50)
`

But its printing:

```
Basic Prearm checks..dont touch!!
Waiting for vehicle to initialize
Arming motors..
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
```

And it keep printing like this, i dont know what Im missing or doing wrong.
And to note i have no gps connected, only PX4Flow and Lidar, thats why Im trying to takeoff in STABILIZE mode.

Any help would be greatful..

Thanking you
Biswajit