dronekit/dronekit-python #632: connect() Dronekit does not work

anbriel, 2016-05-30:
Hi,

I am running the last version of dronekit and pymavlink on my RPI 2, with the v1.1.2 firmware on my pixhawk.
I get this error from the DroneKit example and an immediate timeout (see below).

Using MAVLink 1.0
Connecting to vehicle on: /dev/ttyAMA0
Traceback (most recent call last):
File "vehicle_state.py", line 33, in <module>
vehicle = connect(connection_string, wait_ready=True,baud=57600)
File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 2813, in connect
File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 2086, in initialize
dronekit.APIException: Timeout in initializing connection.

rtfswr, 2016-05-30:
Hey, I get a similar error when connection to SITL on windows (64 bit system), using last dronekit and dronekit-sitl versions, installed via pip on 27.05.2016.

rtfswr, 2016-05-30:
Hey, I solved the problem on my machine following @a7scenario comment --> remove pymavlink 2.0.0, install pymavlink 1.1.69; and it should work.

LorenzMeier, 2016-05-30:
@tridge This sounds like it could be a MAVLink 2.0 related issue.

anbriel, 2016-05-30:
Hi,
I already tried to replace pymavlink 2.0.3 by the 1.1.69 version, but still not working.

peterbarker, 2016-05-30:
On Mon, 30 May 2016, Lorenz Meier wrote:

> @tridge This sounds like it could be a MAVLink 2.0 related issue.

This was related.  dronekit-python was monky-patching a function in 
dronekit - badly.  When that function started receiving keyword arguments, 
dronekit-python started to fail (the dronekit-python function doesn't take 
keyword arguments, so passing thenm in causes failure).

The patch to fix this is already in dronekit-python's head.  No new 
release has been done.

There's currently also an issue some other people are seeing which has 
much the same symptoms.  Somehow people are getting the master version of 
dronekit-python but a release version of SITL.  Haven't gotten to the 
bottom of that yet,

anbriel, 2016-05-31:
Hi,
Here is the screen capture of my raspberry PI2, when I try to run vehicle_state.py.

pi@raspberrypi:~/dronekit-python/examples/vehicle_state $ sudo python vehicle_state.py --connect /dev/ttyAMA0
Using MAVLink 1.0

Connecting to vehicle on: /dev/ttyAMA0
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
>>> Exception in message handler for HEARTBEAT
>>> mode 65536 not available on mavlink definition
Traceback (most recent call last):
  File "vehicle_state.py", line 35, in <module>
    vehicle = connect(connection_string, wait_ready=True,baud=57600)
  File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 2817, in connect
  File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 2168, in wait_ready
dronekit.APIException: wait_ready experienced a timeout after 30 seconds.
Exception in thread Thread-2 (most likely raised during interpreter shutdown):Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 763, in run
  File "build/bdist.linux-armv7l/egg/dronekit/mavlink.py", line 89, in mavlink_thread_out
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'

Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 763, in run
  File "build/bdist.linux-armv7l/egg/dronekit/mavlink.py", line 124, in mavlink_thread_in
  File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 1311, in listener
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'monotonic'

mwingerson, 2016-06-08:
I am also getting the "mode 65536 not available on mavlink definition" error. Any help?

keyurrakholiya, 2016-06-10:
connect doesnt work.
dronekit.APIException: Timeout in initializing connection.
vehicle = connect(connection_string, wait_ready=True,baud=57600)

i am using R-pi with apm. using droneAPI only. not SITL.
any help?
thanks

MikaelFerland, 2016-06-27:
I have the same problem with the PX4 flightstack. I see that dronekit is using the attribute custom_mode to match the actual mode of the autopilot, but this field is a structure here https://github.com/PX4/Firmware/blob/master/src/modules/commander/px4_custom_mode.h#L67.

But it will fail every time here https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L1180 and I think dronekit need a specific case to handle that.

hamishwillee, 2016-07-08:
@peterbarker Did you see this comment? https://github.com/dronekit/dronekit-python/issues/632#issuecomment-228872603

hannahdn, 2016-07-08:
I am having an issue that I think is related. I get a timeout every time I try to connect to my pixhawk since the 2.6.0 release. If I revert back to dronekit-Python 2.5.0 it works fine, using the 1.1.73 version of pymavlink (the 2.0.0 won't allow me to connect, no matter the dronekit version).

janthony, 2016-07-09:
I am trying to connect to 3dr solo. This used to work and now breaking.
Very frustrated with the connection issue. My environment is;

pymavlink==1.1.73
dronekit==2.5.0
monotonic==1.0

I get the following errors;

When I try to connect with 

```
vehicle = connect("udpin:0.0.0.0:14550", wait_ready=True, baud=57600, heartbeat_timeout=30) or
vehicle = connect("udpin:0.0.0.0:14550", wait_ready=False)
vehicle = connect("127.0.0.1:14550", wait_ready=False)

```
```
ERROR LOADING MAVNATIVE - falling back to python implementation
Connecting to udpin:0.0.0.0:14550...
>>> Link timeout, no heartbeat in last 5 seconds
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "helloworld.py", line 10, in <module>
    vehicle = connect(target)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2813, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2086, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.

```

peterbarker, 2016-07-09:
On Thu, 7 Jul 2016, Hamish Willee wrote:

> @peterbarker Did you see this comment? #632 (comment)

I did not, and I think that ticket might be a mashup of a few different 
issues, some of which are fixed :-)

In terms of being ale to connect to a PixHawk using dkpy:

pbarker@bluebottle:~/rc/dronekit-python/examples/vehicle_state(master)$ 
python vehicle_state.py --connect 
/dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00,57600

Connecting to vehicle on: 
/dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00,57600
>>> APM:Copter V3.4-dev (21f2ab51)
>>> PX4: 4a1d6880 NuttX: 579e82d4
>>> Frame: QUAD
>>> PX4v2 0038002D 34324707 37353330

Get all vehicle attribute values:
  Autopilot Firmware version: APM:Copter-3.4.0-dev0
    Major version number: 3

pbarker@bluebottle:~/rc/dronekit-python/examples/vehicle_state(master)$ 
git log -n 1
commit d2db2e71f12d6df717c8bae2becec739aa06dbd7
Author: Ramón Roche <mrpollo@gmail.com>
Date:   Fri Jun 17 10:27:02 2016 -0700

     bump 2.6.0
pbarker@bluebottle:~/rc/dronekit-python/examples/vehicle_state(master)$

pbarker@bluebottle:~/rc/dronekit-python/examples/vehicle_state(master)$ 
python -i -c ""
>>> import pymavlink
>>> print pymavlink
<module 'pymavlink' from 
'/home/pbarker/.local/lib/python2.7/site-packages/pymavlink-2.0.3-py2.7-linux-x86_64.egg/pymavlink/__init__.pyc'>
>>>

So I'm going to need more information before I can reproduce and fix.  A 
vagrant config file would be perfect if someone can arrange it.

janthony, 2016-07-09:
I get the errors when running from the copter instead from my laptop.

peterbarker, 2016-07-13:
@janthony Would you agree that this is a duplicate of #645?

janthony, 2016-07-14:
@peterbarker I can confirm this is a duplicate of #645

peterbarker, 2016-07-22:
@janthony's problem was fixed - that was an issue with Solo.

I believe everyone else's issues here are with the "PX4 Pro" firmware on a PixHawk ( @hannahdn , can you confirm you're running PX4 Pro on your PixHawk?)

MikaelFerland, 2016-07-27:
@peterbarker Sorry for being late on that conversation, but on my side I tested Dronekit with the PX4 Pro / PX4 Flightstack.

JaneClelandHuang, 2016-08-03:
Is there any suggested combination of versions which is known to work reliably?  I am teaching a software engineering class using Iris in a few weeks so am looking for a set of versions for DroneKit-Python, pymavlink, and PixHawk that will work reliably.

I'm having  @a problem with repeated loss of heartbeat after initially establishing a connection:
Connecting to vehicle on: /dev/ttyUSB0,57600
>>> ArduCopter V3.2 (c8e0f3e1)
>>> PX4: dea2cfd1 NuttX: 7a62e351
>>> Frame: QUAD
>>> PX4v2 00200024 32334719 31363032
Connecting to vehicle on:  @/dev/ttyUSB0,57600
>>> Frame: QUAD
>>> Link timeout, no heartbeat in last 5 seconds
>>> ...link restored.
>>> Link timeout, no heartbeat in last 5 seconds
>>> ...link restored.
>>> Link timeout, no heartbeat in last 5 seconds
>>> ...link restored.

hamishwillee, 2016-08-04:
I'm working on Windows against SITL with dronekit 2.8.0, pymavlink 2.0.4 and Copter 3.3.  Haven't tried against Iris in this version.
Assume the connection is USB, then your connection string should be /dev/ttyUSB0 ... I don't think you need to limit to 57600.
That said, the fact that you do get a link restored is indicative that you have some sort of good connection.
@peterbarker @mrpollo  - suggestions for debugging?

majorpr13, 2016-09-18:
Has anyone fixed this issue....I continue to get Exception in message handler for heartbeat/mode 0 not available on mavlink definition. I have tried the most recent and several older versions of dronekit on my companion computer riding with a 3DR solo.

peterbarker, 2016-09-18:
On Sun, 18 Sep 2016, majorpr13 wrote:

> Has anyone fixed this issue....I continue to get Exception in message
> handler for heartbeat/mode 0 not available on mavlink definition. I have
> tried the most recent and several older versions of dronekit on my companion
> computer riding with a 3DR solo.

You will need to use the most recent version of pymavlink, from 
https://github.com/ardupilot/pymavlink

Are you aware of how to do that?

kkroeger007, 2016-09-18:
oh man...I am just noticing this now that the link from a similar issue said to build from the latest pymavlink from mavlink repo which has a redirect on the pymavlink branch to ardupilot pymavlink...apparently that submodule is not the latest. I will pull from here and reinstall and update momentarily.

kkroeger007, 2016-09-18:
@peterbarker So it still has the same issue. To make sure I am doing this correctly.
1) git clone https://github.com/Ardupilot/pymavlink
2) cd into the pymavlink directory
3) python setup.py build install --force --user
is there anything I am missing. When running the connect command with dronekit python I still get 
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
Any help is appreciated.

peterbarker, 2016-09-19:
@kkroeger007 What you're running there looks correct.

... except that that should really have failed, depending on which directory you cloned into!  It looks in various places for a "message_definitions" directory, and you mut have lucked out that it was in one of the locations looked for?

Could you do a "pip uninstall pymavlink" and say which version it is uninstalling?  In fact, you could run that command until it tells you there are not more versions installed *then* run the "python setup.py" line; that *should* ensure you're using the correct one!

JaneClelandHuang, 2016-09-19:
I'm teaching a course right now using Dronekit Python (among other things).  For better or for worse - here are a few course notes that I gave my students - and all twenty of them have the environment set up correctly now. I did tell the students to all use Linux (i.e. native, or VM).  

The notes are an amalgamation of material from websites etc. They aren't really for public dissemination yet, although I plan to post these and other notes once the course is over.
[Setting up DroneKit.pdf](https://github.com/dronekit/dronekit-python/files/479780/Setting.up.DroneKit.pdf)
[Setting up SITL.pdf](https://github.com/dronekit/dronekit-python/files/479781/Setting.up.SITL.pdf)
One of my previous students made a couple of YouTube videos too:
1.     Install SITL/MAVProxy on Linux VM - https://youtu.be/pJGFkZmGV6o 
2.     Install Dronekit for SITL/MAVProxy on Linux VM - https://youtu.be/wSEc7Sm2O0E

kkroeger007, 2016-09-19:
@peterbarker was right...it appeared there was a half installed pymavlink 2.0.6 from somewhere and then a full pymavlink installed over top. So it definitely wasnt grabbing from the right one. I just continued to run pip uninstall pymavlink until it reported that none were installed and then reran the steps I had posted earlier. Thanks for the help. Cleared the issue right up.

peterbarker, 2016-09-21:
Great!  Thanks @kkroeger007 for being clear about the fix.  I'm going to close this issue now on the understanding that dronekit-python works with PX4 Pro firmware.  Let's try to keep it that way...

peterbarker, 2016-09-21:
@JaneClelandHuang Thanks for those links!  On a seemingly-unrelated github issue doesn't seem to be the best place to socialise them; perhaps @hamishwillee might have an idea as to how they might be used.

closed 2016-09-21 by peterbarker

hamishwillee, 2016-09-22:
Firstly, thanks for that Content @JaneClelandHuang . The PDFs aren't particularly needed because we should just have one document for setup, and those should be the official docs. If there are particular flaws that these PDFs fix, then we should discuss those and fix the wiki.

The videos are a different story. We should certainly link to the SITL setup one from the ardupilot wiki. I have created https://github.com/ArduPilot/ardupilot_wiki/issues/555 to track that. 

For Dronekit, perhaps the best solution would be to add a "Video Guides" section [here](http://python.dronekit.io/guide/index.html) and add both videos. Thoughts?

JaneClelandHuang, 2016-09-22:
I have taken down the pdf files (or at least the reference to them).  These were designed as a short-circuit for an entire class of students to get setup quickly at the start of the semester.   Video Guides would be really helpful. I will check with James (the creator of the videos) about making them more public.

deb0ch, 2017-02-19:
**TL;DR:** try `sudo apt-get remove --purge ros-indigo-mavlink` and `sudo pip install pymavlink==2.0.6`

Just a quick feedback on a problem related to this one that kept me blocked for days:

I followed [the tutorials for my ErleCopter on an ErleBrain3](http://docs.erlerobotics.com/simulation/configuring_your_environment) on a fresh install of Ubuntu 14.04 and ran into this precise problem with the same error message.

Trying the solutions proposed here did not work for me until I realized that what the had me install installed the `ros-indigo-mavlink` package, which installs as a dependency `pymavlink` version `1.1.73` (the faulty one). This version conflicted with the one that I installed via pip: even though the `2.0.6` version was installed, python would used ROS' version and I couldn't uninstall it using pip.

The solution was to `sudo apt-get remove --purge ros-indigo-mavlink` and `sudo pip install pymavlink==2.0.6`

tasarmiento, 2017-02-23:
@deb0ch I've tried the multiple pip uninstall pymavlink with a reinstall (so I have version 2.0.6), but am still having connection issues.  Maybe I also have an undiagnosed dependency somewhere.  Is there a way to check for that on my Ubuntu 12.04 VM?