dronekit/dronekit-python #634: AttributeError: 'module' object has no attribute 'start_default'

Tondy, 2016-06-03:
When trying to run sudo python channel_overrides.py

This error appears:
Using MAVLink 1.0
Traceback (most recent call last):
  File "channel_overrides.py", line 18, in <module>
    sitl = dronekit_sitl.start_default()
AttributeError: 'module' object has no attribute 'start_default'

kosovit, 2016-06-06:
I have the same problem .
I use Ubunto
I would like to run the examples in http://python.dronekit.io/ in my droplet ubunto

hamishwillee, 2016-06-06:
Try reinstalling the latest version of sitl from github (and please confirm if it works):

```
sudo pip install git+https://github.com/dronekit/dronekit-sitl
```

arunabha42, 2016-06-07:
Even I am getting the same error. I've followed the install instructions on this page: https://github.com/dronekit/dronekit-sitl#dronekit-sitl
and the example on that page works.

arunabha42@Arunabha-Ubuntu:~/dronekit-scripts$ python hello.py
Start simulator (SITL)
Using MAVLink 1.0
Traceback (most recent call last):
  File "hello.py", line 3, in <module>
    sitl = dronekit_sitl.start_default()
AttributeError: 'module' object has no attribute 'start_default'

arunabha42, 2016-06-07:
Alright, I was having isues upgrading dronekit-sitl from GitHub, I realized that was due to my college proxy. So I've upgraded the dronekit-sitl. However now I'm getting an 
IOError: [Errno 2] No such file or directory:

arunabha42@Arunabha-Ubuntu:~/dronekit-scripts$ python hello.py
Start simulator (SITL)
Starting copter simulator (SITL)
Downloading SITL from http://dronekit-assets.s3.amazonaws.com/sitl/copter/sitl-linux-copter-3.3.tar.gz
Traceback (most recent call last):
  File "hello.py", line 3, in <module>
    sitl = dronekit_sitl.start_default()
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 328, in start_default
    sitl.download('copter', '3.3', verbose=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 161, in download
    return download(system, version, target, verbose=verbose)
  File "/usr/local/lib/python2.7/dist-packages/dronekit_sitl/__init__.py", line 128, in download
    tar = tarfile.open(sitl_target + '/sitl.tar.gz')
  File "/usr/lib/python2.7/tarfile.py", line 1660, in open
    return func(name, "r", fileobj, **kwargs)
  File "/usr/lib/python2.7/tarfile.py", line 1722, in gzopen
    fileobj = bltn_open(name, mode + "b")
IOError: [Errno 2] No such file or directory: '/home/arunabha42/.dronekit/sitl/sitl.tar.gz'

arunabha42, 2016-06-07:
Okay, seems like I've fixed it:
1. I downloaded the sitl-linux-copter-3.3.tar.gz file from http://dronekit-assets.s3.amazonaws.com/sitl/copter/sitl-linux-copter-3.3.tar.gz.

2. Then I put in in my '/home/arunabha42/.dronekit/sitl/' directory.

3. Then I tried running it but that didn't work. So I renamed the file from sitl-linux-copter-3.3.tar.gz to sitl.tar.gz.

That seems to have made it work. Now I've gotten the required output.
What might have caused this issue?

keyurrakholiya, 2016-06-08:
i am facing same problem in r pi. i have replacd the sitl.tar.gz

still getting error : module object has no attribute 'start default'

wilco5nine, 2016-06-08:
Im getting the same issue on mac. Although. Im getting this error in:
- CLI - > ImportError: No module named dronekit . On line 10.

However if run the script in Idle I get;
IDLE ->  AttributeError: 'module' object has no attribute 'start_default' On line 27. 

Im a little confused. Anyhelp would be much appreciated.

dylan93, 2016-06-08:
I get this:

dylan@dylan-VirtualBox:~/dronekit-python/examples/vehicle_state$ python vehicle_state.py
Using MAVLink 1.0
Traceback (most recent call last):
  File "vehicle_state.py", line 27, in <module>
    sitl = dronekit_sitl.start_default()
AttributeError: 'module' object has no attribute 'start_default'

When trying to run "Hello Drone" example. For reference I am running Linux 14.04 on a WM in Windows 8.1.

keyurrakholiya, 2016-06-09:
can i use dronekit-python on raspberry pi? i am facing problem with dronekit - sitl in R-PI.
thanks!

hamishwillee, 2016-06-09:
@dylan93 @wilco5nine That is because we haven't done a new release. Please download DK-Sitl from Github rather than PyPi.

@keyurrakholiya  - downloadable dronekit sitl binaries do not work on RPi - we don't supply ARM binaries. You can build them natively on RPi though and run them from DK-SITL. , Dronkit-Python does run on RPi.

wilco5nine, 2016-06-09:
@hamishwillee I not using PyPi on a RPI, and I cloned the Dronekit-python via git. Im using a Vagrant on a Mac. So im confused. Are the python errors addressed in the new realease? If that is the case great stuff! Ill just wait out then.

hamishwillee, 2016-06-13:
@wilco5nine The problem is probably that you have multiple installations of Python on your machine. The version on CLI is not associated with the PyPi where you installed dronekit-python (which explains that error). The version accessed via IDLE is associated with this PyPi but it is the older version of Dk-Python present.

Fixing the issue with multiple Python installations is something I'd explore via Google :-0. It's not a DK-Python issue really. Last time something like this happened I removed all Python installations and reinstalled.

The error then is because we haven't uploaded a new version of DK-Python to PyPi - this something @peterbarker hopes to do soon. In the meantime, you can reinstall dk-python from github like this:
```
sudo pip install git+https://github.com/dronekit/dronekit-python -UI
```
at that point, your IDLE version will probably work.

AeroCano, 2016-06-21:
Hi, I have the same problem running the examples, i have ubuntu 14.04 and usign MAVLink 1.0, i get:

~/dronekit-python/examples/vehicle_state$ python vehicle_state.py
Using MAVLink 1.0
Traceback (most recent call last):
  File "vehicle_state.py", line 30, in <module>
    sitl = dronekit_sitl.start_default()

I first intalled dronekit following http://python.dronekit.io/guide/quick_start.html.
As i got the error, I verified that I have only one version of pyhton (2.7.6) and I have tried to do all the things you said: I reinstall python (sudo apt-get install --reinstall python2.7) and reinstall python2.7dk-python from github. I still having the same issue, how can i fix it?

Thank you so much.

peterbarker, 2016-06-21:
On Tue, 21 Jun 2016, AeroCano wrote:

> Hi, I have the same problem running the examples, i have ubuntu 14.04 and
> usign MAVLink 1.0, i get:
> 
> ~/dronekit-python/examples/vehicle_state$ python vehicle_state.py
> Using MAVLink 1.0
> Traceback (most recent call last):
> File "vehicle_state.py", line 30, in
> sitl = dronekit_sitl.start_default()
> 
> I first intalled dronekit following
> http://python.dronekit.io/guide/quick_start.html.
> As i got the error, I verified that I have only one version of pyhton
> (2.7.6) and I have tried to do all the things you said: I reinstall python
> (sudo apt-get install --reinstall python2.7) and reinstall
> python2.7dk-python from github. I still having the same issue, how can i fix
> it?
> 
> Thank you so much.

Ramon has done a release of both dronekit-python and dronekit-sitl today.

Could you try updating both of those from the pypy repositories and trying 
again, please?

AeroCano, 2016-06-22:
@peterbarker error changed to this:

:~/dronekit-python/examples/vehicle_state$ python vehicle_state.py
Traceback (most recent call last):
  File "vehicle_state.py", line 13, in <module>
    from dronekit import connect, VehicleMode
ImportError: No module named dronekit

keyurrakholiya, 2016-06-22:
hi, @peterbarker 
i am running all those program successfully.
 use pymavlink version 1.1.73
excute this commands:
`sudo -H pip uninstall pymavlink`
`sudo -H pip install pymavlink==1.1.73`

thank you

peterbarker, 2016-06-22:
On Wed, 22 Jun 2016, AeroCano wrote:

> @peterbarker error changed to this:
> 
> :~/dronekit-python/examples/vehicle_state$ python vehicle_state.py
> Traceback (most recent call last):
> File "vehicle_state.py", line 13, in
> from dronekit import connect, VehicleMode
> ImportError: No module named dronekit

Set the PYTHONPATH environment variable:
export PYTHONPATH=$HOME/dronekit-python

Then run your example again.