dronekit/dronekit-python #635: Send ATT_POS_MOCAP msg through dronekit

pchatrath, 2016-06-06:
Hello,
I want to send motion capture position data to onboard pixhawk on Solo. I understand I need to send ATT_POS_MOCAP  Mavlink signal. However I dont see this msg being supported in the [Copter Commands in Guided Mode](http://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html) documentation. So how do I send the mocap data to solo's pixhawk ?

hamishwillee, 2016-06-06:
@pchatrath The list of supported copter commands is the list from "vanilla" copter 3.3. It is possible that Solo implements this additional command - what makes you think that this is the case? If it does support the command then you can send the message using the approaches described in http://python.dronekit.io/guide/copter/guided_mode.html#guided-mode-commands

I would ask the question about what command is supported on the Solo discussion boards.

pchatrath, 2016-06-07:
Thanks @hamishwillee  I will ask the question on discussion board

closed 2016-06-07 by pchatrath