dronekit/dronekit-python #637: Can we ARMED vehicle using vehicle.overrides?

keyurrakholiya, 2016-06-12:
i am trying to ARM my quadcopter using `vehicle.armed = True`. but its not getting armed.

so i have tried another mathod using vehicle_overrides example.

`print "\nChannel overrides: %s" % vehicle.channels.overrides
print "Set Ch3(throtlle) ch4(yaw) override to 1066 and 1915 (indexing syntax)"

while not vehicle.armed:
    print "arming....."
    vehicle.channels.overrides={'3':110, '4':810}
print "vehicle is ARMED !!!" `

does it right way?

hamishwillee, 2016-06-13:
Hi @keyurrakholiya 

The code fragment above is not enough to understand whether you're doing things right or not. All the [examples in the library](http://python.dronekit.io/examples/index.html) should arm and run properly - at least on SITL. 

Check out the taking off code here: http://python.dronekit.io/guide/taking_off.html

The point is to check `vehicle.is_armable`. IF that value is true then you can arm the vehicle. You then have 10-15 seconds to take off or the system will automatically disarm. The right way to take off is shown in that link.

keyurrakholiya, 2016-06-13:
@hamishwillee 
thank you for your help
i can arm it via RC. but i cant arm it via `vehicle.armed=True`!

i am using R-pi with apm 2.6
and i am not using SITL.

hamishwillee, 2016-06-13:
@keyurrakholiya The arming criteria are the same in both cases, so if you can arm it using RC then it should be good to arm via the command. But it is impossible for me to guess what might be stopping it without all your code ... or even better, you running the takeoff code we provided.

keyurrakholiya, 2016-06-14:
thanks @hamishwillee 
we are using 3.1.5 firmware. should we upgrdge firmware to 3.2.1?

[image]

keyurrakholiya, 2016-06-14:
[image]

hamishwillee, 2016-06-14:
This is sitting on "Waiting to initialise" because is_armable is false. It is not possible to tell why this is from this code. YOu might be able to get some more information by adding the line in the arming check before time.sleep():

```
print "Mode:", vehicle.mode.name
print "GPS...:", vehicle.gps_0.fix_type
```
But chances are that the cause is something that we can't measure, so you should connect to mission planner if you can and see what it says the state is.

> we are using 3.1.5 firmware. should we upgrdge firmware to 3.2.1?

Probably. @peterbarker , `is_armable` is dependent on EKF (`self._ekf_predposhorizabs`). Is there a first version of firmware that we know this can be supported? I have a feeling APM2.6 builds don't turn EKF on by default so this might be a bug?

keyurrakholiya, 2016-06-14:
@hamishwillee  thank you for your help!
i have installed 3.2.1 firmware and its working fine. we tasted autonomous flight simple take off and landing.
thank you!
solved!

hamishwillee, 2016-06-14:
Awesome :-)

closed 2016-06-14 by hamishwillee