dronekit/dronekit-python #638: "dronekit.APIException:" while executing examples/simple_goto.py

koustubh-dwivedy, 2016-06-14:
I am trying to set up dronekit with SITL. I get the following error when I execute simple_goto.py present in the examples dir of dronekit ->
```
Using MAVLink 1.0
Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Connecting to vehicle on: tcp:127.0.0.1:5760
Traceback (most recent call last):
  File "simple_goto.py", line 35, in <module>
    vehicle = connect(connection_string, wait_ready=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2813, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2086, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.
```

What is causing this and how to correct it?

hamishwillee, 2016-06-14:
What platform are you running this on? Can you show the command line you entered and also give the output of `pip list` on your command line?

koustubh-dwivedy, 2016-06-14:
Platform: Ubuntu 14.04.4
`koustubh@koustubh:~/dronekit-python/examples/simple_goto$ python simple_goto.py`
pip list output ->
```
koustubh@koustubh:~/dronekit-python/examples/simple_goto$ pip list
actionlib (1.11.5)
adium-theme-ubuntu (0.3.4)
apt-xapian-index (0.45)
argparse (1.2.1)
backports-abc (0.4)
backports.ssl-match-hostname (3.5.0.1)
bondpy (1.7.16)
bzr (2.7.0.dev1)
camera-calibration (1.12.16)
camera-calibration-parsers (1.11.10)
catkin (0.6.18)
catkin-pkg (0.2.10)
certifi (2016.2.28)
chardet (2.0.1)
colorama (0.2.5)
command-not-found (0.3)
configobj (4.7.2)
cv-bridge (1.11.12)
Cython (0.23.4)
debtagshw (0.1)
decorator (3.4.0)
defer (1.0.6)
diagnostic-analysis (1.8.9)
diagnostic-common-diagnostics (1.8.9)
diagnostic-updater (1.8.9)
dirspec (13.10)
docutils (0.11)
droneapi (1.5.0)
dronekit (2.5.0)
dronekit-sitl (3.0.1)
duplicity (0.6.23)
dynamic-reconfigure (1.5.43)
empy (3.1)
gazebo-plugins (2.4.10)
gazebo-ros (2.4.10)
gencpp (0.5.4)
genlisp (0.4.15)
genmsg (0.5.7)
genpy (0.5.9)
gym (0.1.1, /home/koustubh/gym)
h5py (2.5.0)
html5lib (0.999)
httplib2 (0.8)
image-geometry (1.11.12)
interactive-markers (1.11.1)
Jinja2 (2.7.2)
keyring (3.5)
laser-geometry (1.6.4)
launchpadlib (1.10.2)
lazr.restfulclient (0.13.3)
lazr.uri (1.0.3)
lockfile (0.8)
lxml (3.3.3)
MarkupSafe (0.18)
matplotlib (1.3.1)
MAVProxy (1.4.43)
mercurial (2.8.2)
message-filters (1.11.19)
monotonic (0.6)
netifaces (0.8)
nose (1.3.1)
numpy (1.11.0)
oauth (1.0.1)
oauthlib (0.6.1)
oneconf (0.3.7.14.4.1)
PAM (0.4.2)
paramiko (1.10.1)
pexpect (3.1)
Pillow (2.3.0)
pip (1.5.4)
piston-mini-client (0.7.5)
pluginlib (1.10.2)
protobuf (3.0.0b2)
psutil (4.2.0)
PyBOMBS (2.0.1)
pycrypto (2.6.1)
pycups (1.9.66)
pydot (1.0.28)
pyglet (1.2.4)
Pygments (1.6)
pygobject (3.12.0)
pygpgme (0.3)
pymavlink (2.0.3)
PyOpenGL (3.0.2)
pyOpenSSL (0.13)
pyparsing (2.0.1)
pyserial (3.0.1)
pysmbc (1.0.14.1)
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/list.py", line 80, in run
    self.run_listing(options)
  File "/usr/lib/python2.7/dist-packages/pip/commands/list.py", line 142, in run_listing
    self.output_package_listing(installed_packages)
  File "/usr/lib/python2.7/dist-packages/pip/commands/list.py", line 151, in output_package_listing
    if dist_is_editable(dist):
  File "/usr/lib/python2.7/dist-packages/pip/util.py", line 366, in dist_is_editable
    req = FrozenRequirement.from_dist(dist, [])
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 286, in from_dist
    assert len(specs) == 1 and specs[0][0] == '=='
AssertionError

Storing debug log for failure in /home/koustubh/.pip/pip.log
```

hamishwillee, 2016-06-14:
Try installing a version of pymavlink prior to 2.0.

koustubh-dwivedy, 2016-06-15:
removing pymavlink like `pip uninstall pymavlink` throws `Can't uninstall 'pymavlink'. No files were found to uninstall.` but doing `pip install pymavlink` shows 
```
Requirement already satisfied (use --upgrade to upgrade): pymavlink in /usr/local/lib/python2.7/dist-packages
Cleaning up...
```

hamishwillee, 2016-06-15:
Odd. Maybe try 

```
pip install -I pymavlink==1.1.67
```
Other than that I don't have particular advice on working with pip. You could reinstall dronekit from github but first set that pymavlink version above in the requirements.txt file.

koustubh-dwivedy, 2016-06-15:
I got it working by removing the pymavlink directory in `/usr/local/lib/python2.7/dist-packages` and then downloading pymavlink-1.1.73 from `https://pypi.python.org/pypi/pymavlink/1.1.73` and then running `python setup.py install` in the extracted directory. Now everything is running fine and examples are working.
Thanks for the help @hamishwillee 
I wonder why the latest version is creating problem.

closed 2016-06-15 by koustubh-dwivedy

KisorAlaguvel, 2018-09-23:
@koustubh-dwivedy hey!
I am encountering here the same problem as you were. But I am from Windows platform. Just wanna know whether there is any seperate procedure for this one. 
Also why is that only the simple_goto.py alone experiences this kind of error while takeoff.py works just fine...?

gniewus, 2019-02-03:
I've just experienced a simmilar issue. In my case it was the blocked tcp port 5760, so check if it's free.

i.e on Mac  - `lsof -t -i :5760`

developermastermind07, 2019-05-03:
@koustubh-dwivedy : I am not abe to connect right now .
Can you please provide pymavlink 1.1.73?

SpaceDranzer, 2020-02-10:
Cannot find older pymavlink version. All the pymavlink version on pypi.com are the versions after 2.0. Is there any other way to solve this issue?

tinsmcl1, 2020-07-16:
I am having this same problem, and I cannot find any pymavlink versions before 2.0 either.

GRS26, 2021-01-24:
I am also having this issue, does somebody have this version and could provide?