dronekit/dronekit-python #639: vehicle channels overrides

keyurrakholiya, 2016-06-17:
i want to controll my drone with my laptop keyboard. so i have connected apm to r pi. and r pi has wifi module so further i have connected r pi to my laptop via wifi. now i want to fly my drone with laptop. can i use channel overrides??so i can directly set the value of throttle, yaw ,pitch via my laptop. 

 i have read this here:http://python.dronekit.io/examples/channel_overrides.html
 NOTE: 
Channel overrides (a.k.a "RC overrides") are highly discommended (they are primarily implemented 
for simulating user input and when implementing certain types of joystick control).
 

thank you

hamishwillee, 2016-06-19:
Yes.
Provided you can get the channel over-ride mavlink commands to your ardupilot from your laptop. This is a reasonable use of channel overrides.

closed 2016-06-19 by hamishwillee

keyurrakholiya, 2016-06-20:
thank you @hamishwillee 
can you please tell me the range of values for channel overrides function?
e.g. `vehicle.channels.overrides = {'3':1300}`

can i write between 1300 to 1800 right?

hamishwillee, 2016-06-20:
@keyurrakholiya Depends on what the channel is connected to - your ESC may have certain requirements for its inputs.
YOu can work out what the ranges are set as using the RCx_MIN and RCx_MAX parameters, where x is the channel number. I don't know what they ranges typically are, but I ranges 1000 to 1900 are certainly possible.