dronekit/dronekit-python #640: Setting rate and baud when connecting

alexblack, 2016-06-20:
What values can I use for `rate` and `baud` when calling `dronekit.connect`?  If I set `rate` higher, does that mean I'll have less-stale values for things like Attitude?

https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L2739

hamishwillee, 2016-06-20:
The baud rate is set based on how you're connecting/hardware used. Examples of a few baud rates for some typical connections are given in the docs here: http://python.dronekit.io/guide/connecting_vehicle.html
They will be the same as used for MAVProxy in the same circumstances.

The rate is the rate at which you get *some* data. I'm not sure this includes the positional information - you'd have to check which parameter is actually set under the hood. So yes you could in theory change the rate that information is sent, but my understanding is that this will still be quite limited, and also that some GCS set the value themselves on a regular basis, so even if you set the rate, you may not get to keep that set value.

alexblack, 2016-06-20:
Great, thanks.  I tried rate=20 with sitl and attitude messages came in 5x
more frequently.  I'll test it out on my Solo at some point.

hamishwillee, 2016-06-20:
Great! The stream rate is I believe "per mavlink channel" and probably the channel is reserved for the companion computer.

alexblack, 2016-06-20:
Oh wow, setting `rate=20` doubled the processing time on the drone for anything, I assume thats because DroneKit (or something) is doing a lot more processing to handle that many messages.

hamishwillee, 2016-06-20:
Useful to know.

> I assume that's because DroneKit (or something) is doing a lot more processing to handle that many messages.

I don't know - we'd need profiling to work out what is causing the problem. I wonder if the relationship is linear or not - ie at rate=10 are things half as slow?

Also important to note that even if you can get the messages fast, there is no guarantee that the sensors are updating at that rate or that you can react at a sensible rate ... ie there quite possibly is no benefit at all in real terms in having a much greater message rate.

alexblack, 2016-06-20:
Presumably most sensors update at a frequency higher than 4Hz?

I tried launching an idle dronekit app, that does nothing except connect to a vehicle then sleep, to see what CPU usage of my python process is like:
- At rate=4, cpu usage is 56%
- At rate=10, cpu usage is 92%
- At rate=20, cpu usage is also 92%

92% is the max CPU I seem to be able to get from the drone, the remaining 8% is uses by 3DR processes.

Ah, keep in mind, I've hacked a bit.. damn. I should try to repro this on a clean setup.  I don' let shot_manager on the solo run, so instead I run this code in my process:

```
  mav_vehicle = dronekit.connect("udpout:127.0.0.1:14560", wait_ready=False, source_system=255, use_native=True, heartbeat_timeout=-1)
  out = MAVConnection("udpout:127.0.0.1:14550", source_system=254)
  mav_vehicle._handler.pipe(out)
  out.start()    
```

alexblack, 2016-06-20:
Can I just connect directly using:

```
vehicle = dronekit.connect("udpout:127.0.0.1:14560", wait_ready=False, source_system=255, use_native=True, heartbeat_timeout=-1)
```

Thereby avoiding using `/usr/bin/main.py` or running that code above (*and* connecting to it)?

hamishwillee, 2016-06-20:
@alexblack I don't know sorry. @peterbarker May be able to comment.

alexblack, 2016-06-21:
I tried this out, works great, got me about 1.4x speed improvement on my code.

alexblack, 2016-06-22:
When I take out that extra MAVConnection proxy (that enables shot manager), then, cpu usage is much better, but not great:

* 10% when rate=4
* 13% when rate=10
* 25% when rate=20

tizianofiorenzani, 2017-10-31:
I am facing the very same issue. Setting rate=10 caused my script to slow down so much it was not real time anymore