dronekit/dronekit-python #644: In-door flight with RC_overrides

AndrewOak, 2016-07-07:
I have posted the following question in another issue, by I think it's better to raise a new one.
 
I have been using Dronekit-Python to do RC_overrides for a while, I am now able to arm, take off, hover, and land in a GPS-denied environment. 

I am currently working with Pixhawk v2, and a Raspberry Pi 3B, but I will probably move onto APM2.6 for some other reasons. (the connection is fine, but I'm not sure whether APM2.6 can give the same responses)

My questions are:

1. By lots of experiments, I found out that the minimum time interval between each overrides is 0.393 s. Once set shorter, some values will be ignored. But my project concerns some real-time controls, so is it possible to cut down the time intervals? 
(for example: In the example 'basic mission', we can download, clear, create and upload a mission from/to the flight controller, is it possible to upload a RC_overrides mission onto the flight controller, so the the overrides can be done on board the flight controller? )

2. Since setting RC_channel values is not equal to setting the attitudes (yaw/pitch/roll), a specific override, for example, with channel 2, will cause the UAV to go forward or backward(depending on your value) at a different speed. Can you provide an example concerning the control of movements via setting attitudes(yaw/pitch/roll)?

3. I understand that in a GPS-denied environment, I can not arm in guided mode. Hence, I can not take off using simple_take_off or do any velocity controls. But I am up to modify and compile the source of AC 3.2.1, can you give me some hints to start with. I want to take off using simple_take_off and hover at a desired altitude. My quad has a sonar (MB1240 to be specific), so it should have the necessary feedback to do the trick.

Thank you.

proficnc, 2016-07-07:
My first question would be why you are considering the apm2.6? 
This hardware is not supported in current firmware.
There are many hardware options that use the current generation software, PH1, PH2, PixRacer.... Etc

hamishwillee, 2016-07-07:
@proficnc has a good point. If you use APM2.6 you will be restricted to AC3.2.1 and earlier. This is mostly compatible with AC3.3, but there are some differences - specifically in handling of NED frames and velocity control (which changed from "send velocity command once" to "you have to send velocity commands every couple of seconds or velocity turns off")

hamishwillee, 2016-07-07:
PS Are you SURE that the RC commands need to be re-sent every few seconds?

AndrewOak, 2016-07-09:
@proficnc Actually, my project involves the DIY process of the flight-controller, together with my teammates, we have made two sets of hardware. One based on Pixracer, another one based on APM 2.6. By now, we've only managed to get the latter working. So APM 2.6 is what I get for now. 
Still, if your method relies on firmware beyond AC 3.2.1, and it works well with real-time flight, please tell me how you did it.
@hamishwillee Surely with AC 3.2.1, RC commands don't need to be re-sent every few seconds. Once the override is done, the value of the channel will remain the same.But to navigate the quad in real-time, I think it is necessary that we cut down the time interval. 

So is it possible to upload a RC_overrides mission onto the flight controller, so the the overrides can be done on board the flight controller? Or are there any other options?

hamishwillee, 2016-07-11:
@AndrewOak - No, there is no way to upload RC-Overrides as some sort of mission onto the flight controller. You might be able to get commands handled faster but that would require understanding where the bottlenecks are - I *think* (but do not know) that it is at the pixhawk end and perhaps ardupilot could be configured to have a faster handling loop. I'd discuss that on the ardupilot discussion boards. 
You might also want to ask ardupilot for ETA on "proper" indoor support.

ocastell, 2016-08-31:
I have read this issue and perhaps, I can add some clarifications (or maybe not) ...
I have been writing an App in python (GTK + DroneKit) to control a Drone (quad) in many different ways: using the UI interface, using the keyboard, using a Joystick and with voice commands using julius. With the SITL simulator seems that it works fine ... but in a real drone there are a multitude of variables that can produce a wonderful crash ... if you use the rc_overrides in a real drone then you loose the control from de rc commander and this is disappointing.

My work is in the academic environment and is in progress. You can download and use the App from my repository:

https://github.com/ocastell/DroneControl

I will be happy to get some comments for make it work with a real drone.

hamishwillee, 2016-08-31:
Thanks @ocastell - sounds like it might be interesting. You should cross post this to the discussion server - minimally against [this post](https://discuss.dronekit.io/t/notable-projects-using-dronekit/230/6) but maybe even start a new discussion on the topic of indoor flying.

Biswajit1995, 2017-02-17:
Hello @AndrewOak 

I am also working with dronekit python and using RPi 1 and Pixhawk with copter v3.3.3, and as you I am working in a gps denied environment.

Last few weeks I am trying to make the drone takeoff through python script but unable to do so as i dont know in a gps denied environment, what mode should i use to make the drone takeoff.

And also I want to control the drone through an android mobile app having joystics on screen to send the exact values as of my transmitter and for that I am trying with RC Override but Im not able to takeoff or increase the throttle with rc override as it says `cannot ARM:Throttle too high`.

But first of all, I want to takeoff the drone in GPS Denied environment, hover for few seconds and Land as you have done.

I am using this code for takeoff and land but not sure what mode to use instead of guided,
NOTE: In STABILIZE the motors are disarming after some time instead of increase in their speed:

```
#/usr/bin/env python
# -*- coding: utf-8 -*-
from dronekit import connect,VehicleMode,LocationGlobalRelative
from pymavlink import mavutil
import time
# Receiving from command line
# import argparse
# parser = argparse.ArgumentParser()
# parser.add_argument('--connect',help="PORT_NO")
# args = parser.parse_args()

# Connecting to the vehicle
#connection_string = args.connect
connection_string = "/dev/ttyAMA0,57600"
print("Connecting to...% s" % connection_string)
vehicle = connect(connection_string, wait_ready=True)

#Function to arm and takeoff to a specified altitude
print(vehicle.mode)

# vehicle.mode = VehicleMode("LAND")
# time.sleep(5)
def arm_and_takeoff(aTargetAlt):
	print("Basic Prearm checks..dont touch!!")

 	# while not vehicle.is_armable:
		# print("Waiting for vehicle to initialize")
		# time.sleep(2)

	print("Arming Motors..")
	# Copter should arm in Guided-mode
	vehicle.mode = VehicleMode("GUIDED")
	time.sleep(3)
	vehicle.armed = True
	
	while not vehicle.armed:
		print("Waiting for arming..:% s"% vehicle.mode)
		vehicle.armed = True
		time.sleep(2)
		
	print("Taking Off..")
	vehicle.simple_takeoff(aTargetAlt)
	time.sleep(5)

	while True:
		print("Altitude: ",vehicle.location.global_relative_frame.alt)
		#Break and return from function just below target altitude.
		if vehicle.location.global_relative_frame.alt>=aTargetAlt*0.95: 
			print("Reached Target Altitude..")
			modeLand()
			# print("Landing....")
			# vehicle.mode = VehicleMode("LAND");
			# print("Vehicle mode:%s"% vehicle.mode)
			break

def modeLand():
	print("Landing now")
	vehicle.mode = VehicleMode("LAND")
	time.sleep(5)
	print("-------------Vehicle is in:-------%s"% vehicle.mode)

arm_and_takeoff(2)
print(vehicle.mode)
print("Vehicle object closed")
vehicle.close()
  
```
And I have Optical flow Camera and a Lidar which i can use.
Please help I'm stuck with this.

Regards 
Biswajit

AndrewOak, 2017-04-05:
Hello, @Biswajit1995, I hope you have already overcame this problem.
Sorry for the delay, since I cease using my old mail box for a while.

I'd like to remind you that if your optical flow camera and your lidar is fully functional, you may try to take off in Stabilize Mode with RC_override and switch to loiter mode[1] once you reach your desired altitude. I have not check out yet whether Guided_NoGPS mode is finally implemented or not. Feel free to check it out as well.

[1]http://ardupilot.org/copter/docs/loiter-mode.html#loiter-mode

For the range of the RC values, check out those answers by jmachuca77 in this post:  https://github.com/dronekit/dronekit-python/issues/643#issuecomment-291568331

Good luck and be careful!

Biswajit1995, 2017-04-27:
Hey @AndrewOak ,

Thank you for your reply, and this time sorry for my late reply. Since I was busy in trying something else.

As you have mentioned, I have already overcome my problem that i was having with channel overrides and built an android controller for my quad. 

But now a new issue arises, as the next task I want to do is to Takeoff the quad upto a height, and then i have to make my quad follow a line on the ground and land at the line end. For this I am planning to use OpenCV, but not sure after detecting the line, how will I make the quad follow it. 

But before that, I want to use GUIDED_NOGPS, and for that I have updated my Pixhawk firmware to copter 3.4.6, and suddenly it stopped detecting my RC in RC calibration tab. I have updated multiple times, but got no success. In calibration window of Mission planner no Green bars are showing up, and similarly in QGroundControl, it is saying, "**no RC channels detected, need at least 5 channels for PX4**" 

I am using Futaba T14SG transmitter and Futaba R7008sb SBUS2 receiver, the same RC that i was using with previous copter version, and it was working fine and still working fine with copter v3.3. Don't know why it is happening with copter v3.4.6.

I have searched many blogs and forums, posted this issue many times but got no responses till now.

If you can help me this, that would be really helpful.

Thanks
Biswajit1995

toashwin, 2017-05-03:
@Biswajit1995 did you make any progress?
Not sure if you can share your code on controlling the drone with rc_override?

I also need a solution like yours.. follow a line using a camera. if you get lucky let me know. Came across this one.. which is using openMV
http://diydrones.com/profiles/blog/show?id=705844%3ABlogPost%3A2369576&commentId=705844%3AComment%3A2369680&xg_source=activity

Thanks

AndrewOak, 2017-05-03:
@Biswajit1995 Good day,

For your Problem No.1: 

- If you are switching to a simple camera plus openCV solution, my answer would be to do it the dumb way: First you may define some pixel position, say [320, 240] if your camera is 640×480, in your field of view as the current position of your drone. If the line is detected on your right, change you roll value (with caution) like the way you control the drone with your RC transmitter, the vice versa. Remember to set a zone of tolerance, which means if the line is not too right or too left, tell the drone to "consider it done" and you may change the pitch value to tell it go forward or backward during this period.
- If you are using a camera, a optical flow sensor (possibly the same camera ? ), and a lidar, you can still do it the dumb way (which can already be a huge success after some debugging). But how about study your lidar or optical flow sensor more closely, understand how they are working with your flight control, what value are they sending to your flight control board ? If you are capable of editing and compiling your DIY version of your firmware, how about enabling it to handle the information you retrieved with your camera ? If not, how about send the information in the same way your optical flow sensor or Gps module send them ? So your flight control board can do it the smarter way rather than simply following the orders of their master?

For your problem No.2:

I regret that I don't have anything to test the new firmware, but if it is not your problem, Google it, or ask the community. If it is your problem, reset, reinstall everything and start again. I think you may already overcome this problem by yourself ：)

Best regards !

Biswajit1995, 2017-05-24:
Thanks @
For the suggestions, even I am thinking of sending the quad attitude messages instead of RC overrides to move the quad. But now stuck at, taking off the drone in GUIDED_NOGPS mode.
Im following this link [Movement using GUIDED_NOGPS](https://github.com/dronekit/dronekit-python/pull/712/files/61e4dbda8693128133d488048f52a5187129f208#diff-d5da2834b559fea34f0a96f46aa3387c) got no success. @
Check if this can help you with the project.

And for the Problem no 2. I have solved the issue as the problem was with my receiver as it is supporting only SBUS2, but from copter 3.4.+ it has to be SBUS and SBUS2 is depreciated from the current copter v3.4+.

Regards
Biswajit

kevinpallado, 2019-08-21:
@AndrewOak, @Biswajit1995  do you successfully control your plane using python script? Do you have any information/guide/steps on how to fully control it manually. I am planning to control a plane from my keyboard with the help of python script. I am not quite good in drones, I understand a little bit but in programming theres is no problem. Hope you can help me