dronekit/dronekit-python #645: vechicle.connect() timeout

janthony, 2016-07-09:
I am trying to connect to solo with below code;

```
vehicle = connect("udpin:0.0.0.0:14550", wait_ready=True, baud=57600, heartbeat_timeout=30) or
vehicle = connect("udpin:0.0.0.0:14550", wait_ready=True)
vehicle = connect("udpin:0.0.0.0:14550", wait_ready=False)
```

My environment is

```
pymavlink==1.1.73
dronekit==2.5.0
monotonic==1.0
```

I get the following errors when running on the solo

```
ERROR LOADING MAVNATIVE - falling back to python implementation
Connecting to udpin:0.0.0.0:14550...
>>> Link timeout, no heartbeat in last 5 seconds
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "helloworld.py", line 10, in <module>
    vehicle = connect(target)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2813, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 2086, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.
```

franknitty69, 2016-07-09:
@janthony, the error you are seeing means that whatever is invoking the script cannot connect to solo on udpin:0.0.0.0:14550. where are you running the script from?

my scripts running on my laptop connect to solo just fine.

janthony, 2016-07-09:
My script runs fine from my laptop as well. But when I run from the solo
computer, then it fails everytime.

franknitty69, 2016-07-10:
@janthony i can confirm. the same script that runs on the computer connects to solo and runs just fine. if i move that script to solo and run it, the script can't connect and times out. the error occurs if i run the script directly via python script.py or via solo script pack.

**via solo script run**
Exception in thread Thread-2 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", line 166, in mavlink_thread_out
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", line 201, in mavlink_thread_in
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init__.py", line 1311, in listener
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'monotonic'

peterbarker, 2016-07-11:
On Sun, 10 Jul 2016, Ahmed Agbabiaka wrote:

> @janthony i can confirm. the same script that runs on the computer connects
> to solo and runs just fine. if i move that script to solo and run it, the
> script can't connect and times out. the error occurs if i run the script
> directly via python script.py or via solo script pack.
> 
> via solo script run
> Exception in thread Thread-2 (most likely raised during interpreter
> shutdown):
> Traceback (most recent call last):
> File "/usr/lib/python2.7/threading.py", line 551, in bootstrap_inner
> File "/usr/lib/python2.7/threading.py", line 504, in run
> File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py",
> line 166, in mavlink_thread_out
> : 'NoneType' object has no attribute 'error'
> Exception in thread Thread-1 (most likely raised during interpreter
> shutdown):
> Traceback (most recent call last):
> File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
> File "/usr/lib/python2.7/threading.py", line 504, in run
> File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py",
> line 201, in mavlink_thread_in
> File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/__init.py",
> line 1311, in listener
> : 'NoneType' object has no attribute 'monotonic'

OK, so sadly these exceptions don't indicate the source of your issues.

Note the message "during interpreter shutdown", and "'Nonetype' 
object..."; the "None" objects there are python modules.  These errors 
occur because the main thread is busy deconstructing itself and the 
interpreter, but the mavlink threads are still running.

These threads should *really* be shut down nicely before we quit, but 
aren't.

Nobody has gone and fixed that as by the time you get them your program 
has finished and it is just messy.  OTOH, it does lead to distractions 
from the real issue - like the one you're facing.

Could you give me the exact incantation of solocli you're using, please?

Peter

janthony, 2016-07-11:
@peterbarker Thanks for taking your time to reply. The command I run from my computer is

`solo script run helloworld.py`

To short circuit the debug times, I tried ssh into solo and cd into the script dir and run the `python helloworld.py` from there. 

Both fail for me. I am trying to figure out a way to debug the issue on the solo itself. If you have any pointers as to how I can go about debugging, I could dig into this deeper.

peterbarker, 2016-07-12:
On Sun, 10 Jul 2016, Jerome Anthony wrote:

> @peterbarker Thanks for taking your time to reply. The command I run from my
> computer is
> 
> solo script run helloworld.py

OK, so that one runs fine for me.  Old version of Solo's software, 
'though.  I'll update my solo and see if something broken in the recent 
Solo releases.

> Both fail for me. I am trying to figure out a way to debug the issue on the
> solo itself. If you have any pointers as to how I can go about debugging, I
> could dig into this deeper.

What does "python helloworld.py" do for you?

peterbarker, 2016-07-12:
On Sun, 10 Jul 2016, Jerome Anthony wrote:

> @peterbarker Thanks for taking your time to reply. The command I run from my
> computer is
>
> solo script run helloworld.py
>
> To short circuit the debug times, I tried ssh into solo and cd into the [...]

After updating my Solo, I now get a timeout with helloworld.py

3DR may have broken something with telem_forwarder - I no longer have 
access to that code.

What *does* work is this:

diff --git a/examples/helloworld/helloworld.py 
b/examples/helloworld/helloworld.py
index b51d02c..89b403d 100644
--- a/examples/helloworld/helloworld.py
+++ b/examples/helloworld/helloworld.py
@@ -2,7 +2,8 @@ from dronekit import connect
  import sys

  # Connect to UDP endpoint (and wait for default attributes to accumulate)
-target = sys.argv[1] if len(sys.argv) >= 2 else 'udpin:0.0.0.0:14550'
+#target = sys.argv[1] if len(sys.argv) >= 2 else 'udpin:0.0.0.0:14550'
+target = sys.argv[1] if len(sys.argv) >= 2 else 'udpout:10.1.1.10:14560'
  print 'Connecting to ' + target + '...'
  vehicle = connect(target, wait_ready=True)

I'm wondering if 3DR might have nixed those broadcasts on Solo's localhost 
interface or something.....

See if that patch works for you?

janthony, 2016-07-12:
@peterbarker Thanks. Will test this and get back to you. I thought this connection string you specified,
`udpout:10.1.1.10:14560`
 is what is used when you are proxying through MapProxy

franknitty69, 2016-07-12:
@peterbarker i am running
solo-cli 1.2.0
dronekit 2.7.0
solo fw 2.4.0

if i run helloworld from solodevguide/examples via solo script run, the script runs (**udpin:0.0.0.0:14550**) and i can see all of the vehicle attributes. but on vehicle.close, the script doesn't exit graciously.

janthony, 2016-07-12:
@franknitty69 I installed solo-cli 1.2.0. But when I try `solo --version` it says version `solo-cli 1.1.2`. Does yours print `solo-cli 1.2.0` ?

franknitty69, 2016-07-12:
if i use pip list, solo-cli is 1.2.0. if i use solo --version it says 1.1.2. i'm not sure where the discrepancy is.

peterbarker, 2016-07-12:
On Mon, 11 Jul 2016, Jerome Anthony wrote:

> @peterbarker Thanks. Will test this and get back to you. I thought this
> connection string you specified,
> udpout:10.1.1.10:14560
> is what is used when you are proxying through MapProxy

That's a "pymavlink-style" connection string.  It is passed through to 
pymavlink pretty much unmolested by both dronekit-python and mavproxy.

The thing listening on Solo's :14560 is *not* mavproxy IIRC - it's 
telem_forward.

janthony, 2016-07-12:
@peterbarker I reset my solo/controller and then the connection timeout went away. I believe the problem was after the software update I had not paired the solo to the controller (strangely I could control the gimbal using the controller, which contradicts this theory).

But I get the following error of script not ending successfully;

```
ERROR LOADING MAVNATIVE - falling back to python implementation
Using MAVLink 1.0
Connecting to udpout:10.1.1.10:14560...
>>> PreArm: Waiting for Nav Checks
>>> APM:Copter solo-2.0.20 (609645b5)
>>> PX4: bf80a499 NuttX: a5146c68
>>> Frame: QUAD
>>> PX4v2 0034001C 31355104 36333436
>>> PreArm: Waiting for Nav Checks
>>> PreArm: Waiting for Nav Checks
Vehicle state:
 Global Location: LocationGlobal:lat=xxx,lon=xxx,alt=96.86
 Global Location (relative altitude): LocationGlobalRelative:lat=-xxx,lon=xxx,alt=1.66
 Local Location: LocationLocal:north=None,east=None,down=None
 Attitude: Attitude:pitch=-0.0601995512843,yaw=-0.326566100121,roll=-0.010718235746
 Velocity: [-0.01, 0.01, -0.01]
 Battery: Battery:voltage=15.796,current=0.7,level=74
 Last Heartbeat: 0.856388333
 Heading: 341
 Groundspeed: 0.0399999991059
 Airspeed: 0.0399999991059
 Mode: LOITER
 Is Armable?: True
 Armed: False
>>> PreArm: Waiting for Nav Checks
>>> PreArm: Waiting for Nav Checks
>>> PreArm: Waiting for Nav Checks
>>> PreArm: Waiting for Nav Checks
Done.
Exception in thread Thread-2 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run
  File "/log/solo-script/env/lib/python2.7/site-packages/dronekit/mavlink.py", line 166, in mavlink_thread_out
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'error'
Exception in thread Thread-1 (most likely raised during interpreter shutdown):
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 551, in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 504, in run

```

This happens both with `udpout:10.1.1.10:14560` and `udpin:0.0.0.0:14550`

peterbarker, 2016-07-12:
On Tue, 12 Jul 2016, Jerome Anthony wrote:

> But I get the following error of script not ending successfully;

Yes, those threads aren't terminated correctly.  I think I have to do 
something about that  - somehow.  Last time I looked it was complicated 
due to being daemon threads.

janthony, 2016-07-12:
I can contribute to debugging and fixing if you can give a few pointers as to how to proceed. 
On a side note, if you pair the solo, again with the controller, you might be able to get it connecting via the app.

peterbarker, 2016-07-13:
@janthony This branch: https://github.com/dronekit/dronekit-python/compare/master...peterbarker:stop-threads?expand=1 contains sufficient to stop those threads from dying nasty, horrible deaths.

Do you think you could test that one for me, as it seems you had the interpreter-shutdown error happening quite reliably.

janthony, 2016-07-14:
@peterbarker Sure. I will test it and let you know

janthony, 2016-07-16:
I have tested this and it works. Thanks for the fix.

franknitty69, 2016-07-18:
i tested the fix as well and can confirm that dronekit-python scripts now exit gracefully.

peterbarker, 2016-07-22:
This was merged in - thanks @franknitty69!

peterbarker, 2016-07-22:
Note that the original issue here - timeout in connect() on Solo - was solved by @janthony with a reset of both Solo and Artoo.

closed 2016-07-22 by peterbarker