dronekit/dronekit-python #647: Possible to set pitch/roll in the way that we set yaw?

AndrewOak, 2016-07-15:
Hello there, I'm still trying to navigate my quad in a GPS-denied environment.
Though I've managed to take off, hover, and land with RC_override, I finally came to the conclusion that this method is just not reliable, and I'd like to try something else.

By browsing the documentation, I understand that it is possible to set the yaw in the following method:
http://python.dronekit.io/guide/copter/guided_mode.html#guided-mode-copter-set-yaw

So can we set roll/pitch in some similar way?
I discovered that we can set the ROI of the camera like this: 
http://python.dronekit.io/guide/copter/guided_mode.html#guided-mode-copter-set-roi
Does it work with the vehilcle itself?

Above all, is it possible to set/control the attitude of the vehicle with Dronekit Python in a GPS denied environment?

hamishwillee, 2016-07-18:
Hi @AndrewOak,

True, GPS is currently fundamental to many ardupilot modes. 

The simple answer is I don't know what you can do to get around this - but you can find out what commands were supported in the last version of ardupilot here: http://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html
I experimented with the ROI at some point. How this works depends on the type of gimbal (ie it points the vehicle if you don't have a gimbal on board. But I don't know about GPS dependencies for this.

The longer answer is that this isn't really a dronekit question - we just send commands to the autopilot. I think you're best off raising these discussions in the discussion forums, and then if you find a message you know is required, requesting it's addition (or calling it yourself, which DK supports).

Regards
Hamish

agribov, 2016-08-21:
Hey @AndrewOak , have you made any more progress since these two posts? I'm looking into using DroneKit to operate in a GPS deprived environment, and was curious if you managed to set the Roll/Pitch values. Looking through the documentation, it seems that the [velocity control ](http://python.dronekit.io/guide/copter/guided_mode.html#velocity-control)feature does what we're looking to do.

bkakilli, 2016-10-20:
Hi @AndrewOak and @agribov. Have you made any further progress about indoor drone controlling with DroneKit and Pixhawk? I've tried Guided_NoGPS and RC override as you did, but I couldn't manage to take off. Does anyone think Pixhawk and DroneKit platform is not the correct choice for that purpose? I think they are capable though.

agribov, 2016-10-20:
Ended up giving up on Dronkekit, didn't seem developed enough for my purposes, and there wasn't much support. Instead I'm passing the sbus signal through another microcontroller on the way to the FC, and having that microcontroller run my own proprietary code.

mikeklinker129, 2017-05-09:
There is a Mavlink message to set the attitude target. SET_ATTITUDE_TARGET should be implemented for arducopter 3.4+