dronekit/dronekit-python #659: Call to upload() Doesn't Always Return

ben24, 2016-09-01:
I have the following problem that is easy to reproduce on Ubuntu.  Version of Dronekit is 2.4.0

I isolated problem with debugger.  Function is below and snippet that hangs is 

        "while False in self._vehicle._wp_uploaded:
            time.sleep(0.1)"

def upload(self):
    if self._vehicle._wpts_dirty:
        self._vehicle._master.waypoint_clear_all_send()
    if self._vehicle._wploader.count() > 0:
        self._vehicle._wp_uploaded = [False] * self._vehicle._wploader.count()
        self._vehicle._master.waypoint_count_send(self._vehicle._wploader.count())
        while False in self._vehicle._wp_uploaded:
            time.sleep(0.1)
        self._vehicle._wp_uploaded = None
    self._vehicle._wpts_dirty = False

kevinjhur, 2017-03-06:
I have same issue.
https://github.com/dronekit/dronekit-python/issues/700

Did you solve this problem?

tizianofiorenzani, 2019-02-07:
Just got this issue sometime:
/dronekit/__init__.py", line 3024, in upload

TypeError: argument of type 'NoneType' is not iterable

peterbarker, 2019-03-13:
@tizianofiorenzani Quick look seems to say you can get this is if you call upload() twice from different threads.  Any chance you did this?

tizianofiorenzani, 2019-03-13:
I don't think so, but I need to double check. Looking at the code, the only way _wp_uploaded becomes None is after the while loop. Before that, is initialized as a boolean array. The least it can be is []. So your explanation is sound