dronekit/dronekit-python #664: Camera support?

antonvh, 2016-09-12:
I'm coding camera control and geotagging. Would you like to add camera classes to dronekit? Would you like a PR?

Here's my first working code: https://github.com/antonvh/dronekit-python/blob/gopro_wifi/dronekit/cameras.py

hamishwillee, 2016-09-13:
Thanks @antonvh 

@mrpollo @peterbarker These are architectural questions, best answered by you guys. Ideally we'd provide a pluggable-camera API. 

Even if we can't include now, we should add to the [Notable projects using DK](https://discuss.dronekit.io/t/notable-projects-using-dronekit/230)

mrpollo, 2016-09-14:
@antonvh why don't you submit the PR?, I think it would be a great addition to dronekit, once you submit a PR I'll give you some feedback

antonvh, 2016-09-14:
PR done: https://github.com/dronekit/dronekit-python/pull/665