dronekit/dronekit-python #675: Can't run Dronekit examples!!

SalahuddinAsh, 2016-10-20:
Hi everyone, 

I'm trying to run the examples, just to get a rough understanding of DroneKit.
I've followed the [Running the Examples](http://python.dronekit.io/examples/running_examples.html) guide. When I run 
`python vehicle_state.py`
it gives me errors
`Starting copter simulator (SITL)
SITL already Downloaded and Extracted.
Ready to boot.
Traceback (most recent call last):
  File "guided_set_speed_yaw.py", line 33, in <module>
    sitl = dronekit_sitl.start_default()
  File "/home/salahuddin/.local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 341, in start_default
    sitl.launch(sitl_args, await_ready=True, restart=True)
  File "/home/salahuddin/.local/lib/python2.7/site-packages/dronekit_sitl/__init__.py", line 271, in launch
    p = Popen([self.path] + args, cwd=wd, shell=sys.platform == 'win32', stdout=PIPE, stderr=PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 8] Exec format error
`

And if I tried to run using --connect like this
`python guided_set_speed_yaw.py --connect tcp:localhost:5760`
it works for a while and stuck at infinitely displaying `Waiting for arming ...` and sometimes display 
`>>> PreArm: RC not calibrated
>>> PreArm: Compass not calibrated
>>> PreArm: Accels not calibrated
`
I'm a totally newbie with DroneKit. I appreciate your help to get over this starting problem 

Many thanks

peterbarker, 2016-10-21:
Because you're starting the simulated vehicle by hand (rather than using a
pre-canned dronekit-sitl binary), you will need to set some parameters on
the vehicle before being able to arm.  The ArduPilot binary takes a
"--defaults" option, which you should point to
Tools/autotest/default_params/copter.parm (this file has moved in recent
history; it might be one level up).

hamishwillee, 2016-10-22:
@peterbarker ... but the `python vehiclestate.py` should just work out of the box.

franknitty69, 2016-10-22:
i just ran guided_set_speed_yaw.py and vehicle_state.py with DroneKit-SITL 3.2.0 and DroneKit 2.9.0 and they ran just fine.

are you using the latest version of dk-sitl?

SalahuddinAsh, 2016-10-22:
Many thanks for you all guys for your time.
@peterbarker I just started as instructed in the documentation, which says nothing about setting parameters. Could someone add this to the documentation?
When I try running `dronekit-sitl ~/ardupilot/ArduCopter/ArduCopter.elf` it just works fine and wait for connection.
When I try running `python vehicle_state.py --defualts ~/ardupilot/Tools/autotest/default_params/copter.parm` it says .. 
`vehicle_state.py: error: unrecognized arguments: --defualts /home/salahuddin/ardupilot/Tools/autotest/default_params/copter.parm
`
When I try running `python vehicle_state.py --connect tcp:localhost:5760`, it also works and start connecting the sitl.
But I simply need to run  the example to get the output stated in the documentation. How can I do it and how can I use the --defaults option?
@peterbarker I expected it to work out of the box as there is nothing about special configuration in the Dronekit documentation.
@franknitty69 I'm using the latest versions. Here is the output of installing the tools
`Successfully installed dronekit-2.9.0 dronekit-sitl-3.2.0 monotonic-1.2 psutil-4.3.1 pymavlink-2.0.6 six-1.10.0`

Thanks again

hamishwillee, 2016-10-22:
Howdy, I've tested this heaps of times, I wrote that documentation, and you should NOT need to do anything other than what is stated in the instructions. I suspect there is something broken in your DK-SITL installation.

W.r.t. the defaults issue, note you are miss-spelling it "defualts"

It is interesting that `python vehicle_state.py --connect tcp:localhost:5760` works.

SalahuddinAsh, 2016-10-22:
@hamishwillee Thanks for your reply. Sorry for the typo in defaults, I've tried it with the right spelling but it tells me that 

`python vehicle_state.py --defaults ~/ardupilot/Tools/autotest/default_params/copter.parm 
usage: vehicle_state.py [-h] [--connect CONNECT]
vehicle_state.py: error: unrecognized arguments: --defaults /home/salahuddin/ardupilot/Tools/autotest/default_params/copter.parm`

How can I know what is broken in DK-SITL and how to fix it. I re-installed it many times by the way. 
I appreciate your help

franknitty69, 2016-10-22:
i think something is wrong with your DK-SITL as your issue is isolated.

try re-installing dk sitl.

sudo pip uninstall dronekit-sitl

sudo pip install dronekit-sitl

SalahuddinAsh, 2016-10-22:
@franknitty69 Thanks for your help. I've tried to follow your suggestion. Uninstall then re-install DK-SITL but nothing changed. I've even tried to install it on another machine but nothing changed. What do you think?
Could someone try to install DK-SITL now using sudo pip install dronekit-sitl. There may be something wrong with the available release. I don't know, may be!!

SalahuddinAsh, 2016-10-23:
Guys, no more suggestion to solve this problem??

hamishwillee, 2016-10-23:
Hi @SalahuddinAsh ,

I just tried this on vanilla Ubuntu 16.04 following the basic instructions (I did have to install git, which is not present by default). All worked fine.

```
sudo apt-get install python-pip python-dev
pip install dronekit
pip install dronekit-sitl
supo apt install git
git clone http://github.com/dronekit/dronekit-python.git

cd dronekit-python/examples/vehicle_state/
python vehicle_state.py

ubuntu@ubuntu:~/dronekit-python/examples/vehicle_state$ python vehicle_state.py Starting copter simulator (SITL)
Downloading SITL from http://dronekit-assets.s3.amazonaws.com/sitl/copter/sitl-linux-copter-3.3.tar.gz
Download Complete.
Payload Extracted.
Ready to boot.

Connecting to vehicle on: tcp:127.0.0.1:5760
>>> APM:Copter V3.3 (d6053245)
>>> Frame: QUAD
>>> Calibrating barometer
>>> Initialising APM...
>>> barometer calibration complete
>>> GROUND START

Get all vehicle attribute values:
 Autopilot Firmware version: APM:Copter-3.3.0
   Major version number: 3
   Minor version number: 3
   Patch version number: 0
   Release type: rc
   Release version: 0
   Stable release?: True
...
```

So I'm not sure what to suggest.

peterbarker, 2016-10-24:
@SalahuddinAsh Note that I said that the *ArduPilot* binary has a "--defaults" option, not that the vehicle_state.py script has a "--defaults" option :-)

SalahuddinAsh, 2016-10-24:
@hamishwillee Good news. I've tried to boot from a live USB and then "Try Ubuntu". When I installed Dronekit and tried the examples it worked without any troubles. I guess there is some thing wrong (may be conflict between some installed packages on my PC. What do you think? How can I fix it? I think there are ways better than reinstalling Ubuntu, right?

@peterbarker I'm sorry but I can't understand the relationship between running Dronekit examples and ardupilot binary? Why should I point to something in ardupilot project to get the Dronekit examples to work? Could you please clarify this to me? And tell me HOW to use the "--default" option?

Many thanks

hamishwillee, 2016-10-24:
Hi @SalahuddinAsh 

I haven't tried it, but perhaps you could install [virtualenvwrapper](https://virtualenvwrapper.readthedocs.io/en/latest/) and try run dronekit-sitl from in there. That would at least reduce chance of conflicting packages.

The RPi problem is because Dronekit only supplies X86 binaries, not ARM binaries (as required by RPi). Running on RPi isn't one of the supported use cases. If you wanted to use SITL on RPi you could, but you'd have to build the binaries yourself. Dronekit SITL can then be used to launch the ardupilot binary you build.

I don't know exactly what @peterbarker has suggested. My guess though is that he is saying that if DK-SITL doesn't work, you can still build SITL natively. By default the examples start and connect to the address used by DK-SITL, but you can start SITL yourself and connect to it yourself (syntax in the docs).

SalahuddinAsh, 2016-10-24:
Hi @hamishwillee. Thanks for your time. 

I've installed a new Ubuntu 16.04 on my home-device and Dronekit's examples are working fine now. Maybe I'll try virtualenwrapper or reinstall Ubuntu on my work-device too. 

Regarding the RPi, I think that DroneKit supports RPi as a companion computer as mentioned in the Dronekit documentation [here](http://python.dronekit.io/develop/companion-computers.html) and also on the ardupilot's wiki [here](http://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html). 
There is also one guy "Dennis" who has a video on [Youtube](https://www.youtube.com/watch?v=7QDb2SuUVsw) on how to use Dronekit on RPi by following the instructions on the previous two links. 

What do you think? And could you please tell me how to build Dronekit myself?

hamishwillee, 2016-10-24:
@SalahuddinAsh What those documents means it that DroneKit-Python will run on RaPi. That makes sense, because you're using DK-Python to remotely control the autopilot from the companion computer.

There is no mention that you can run **DroneKit-SITL** on Ra-Pi. We don't support it, though we could by building the ARM binaries. This isn't a priority for us - we don't see great need to simulate ardupilot on an RPi

>  And could you please tell me how to build Dronekit myself?

I suspect that you are actually asking how to build ArduPilot so that you can call it from DroneKit-SITL (or launch it yourself)?

The docs on this are here: http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html#sitl-simulator-software-in-the-loop

I am not sure how reliable they are, because ArduPilot just updated its build system. 

I think you may be confused. I think you probab

SalahuddinAsh, 2016-10-24:
Thanks @hamishwillee for your quick reply and sorry for the further questions.
Actually I'm not trying to run DroneKit-SITL on RPi. I want simply to use DroneKit to send simple command from RPi to Pixhawk (e.g., climb up, down, yaw right, left). 
I thought the best way to start learning how to do that is by going through DroneKit's examples, right? However, when I tried to run the examples in RPi they didn't work and gave me the same errors as this issue.
I'm sorry if I'm confused. When I skimmed the examples if found that they are using sitl, right?
Could you please clarify me these points and the best way to learn how to reach my goal.

Many thanks :)

hamishwillee, 2016-10-25:
Hi @SalahuddinAsh 

It goes like this. DK-Python will run on almost any platform that runs platform - including RPi. 

DK-SITL is a framework that launches ArduPilot SITL binaries. While DK-SITL itself will run on almost any platform, we only supply ArduPilot binaries for x86 platforms - so basically your Windows 10, Mac OS X or Linux PC using Intel/AMD chips. If you want to use DK-SITL on ARM platforms like RPi it won't be able to download the binaries. YOu can however build SITL natively to create the needed binaries and then DK-SITL can launch those.

The examples use DK-Python to connect to a real or simulated vehicle using a specified "connection" for where on the network the vehicle is located. If you don't specify any connection then the examples use DK-SITL to launch SITL and connect to it.

If you want to start testing on RPi, then you would have to run SITL on another machine on the same network, then connect the RPi to that. 

For your case, I would step away from your RPi altogether and run the examples on a desktop. When you're happy with the behaviour, push the scripts to RPi and run them talking to your autopilot.

Hope that helps.

SalahuddinAsh, 2016-10-25:
Million thanks @hamishwillee, this really helped me understand how it works. 
I'll start, as you suggested, with running the examples on the pc, then moving to the RPi. 
Would you mind ask you on Gitter if I have further questions in the next days.

hamishwillee, 2016-10-25:
Feel free to ask questions on gitter - in fact that is better. But address them generally so that more people can help. I'm away for a lot of the time.

closed 2016-10-25 by hamishwillee

peterbarker, 2016-10-25:
On Thu, 20 Oct 2016, SalahuddinAsh wrote:

> And if I tried to run using --connect like this
> python guided_set_speed_yaw.py --connect tcp:localhost:5760
> it works for a while and stuck at infinitely displaying Waiting for arming ... and [...]

Because you're starting the simulated vehicle by hand (rather than using a 
pre-canned dronekit-sitl binary), you will need to set some parameters on 
the vehicle before being able to arm.  The ArduPilot binary takes a 
"--defaults" option, which you should point to 
Tools/autotest/default_params/copter.parm (this file has moved in recent 
history; it might be one level up).