dronekit/dronekit-python #677: Exception in message handler for HEARTBEAT    mode 0 not available on mavlink definition

patrickpoirier51, 2016-10-24:
Hello

Using dronekit 2.9.0 (confirmed within python)

I am running a simple test from my OdroidXU4 on ubuntu 14.04
MINIMAL_TEST.py
from dronekit import connect
import time
print 'Connecting to vehicle on: %s' % connection_string
vehicle = connect('tcp:my.host.computer.ip:5763', wait_ready=True)
while True:
print " Ch1: %s" % vehicle.channels['1']
time.sleep(0.2)

I do NOT have error message when connecting to my UBUNTU Host running sitl : sim_vehicule.py 

I do have error message when connecting to my WINDOWS Host running simulator within Mission Planner 1.3.41
I do have error message when connecting to WINDOWS Mission Planner using mavproxy --master tcp:host.windows.host.pc:5763 --out udp:127.0.0.1:14550  on the Odroid  and connecting dronekit with @0.0.0.1:14550

Please let me know if you require additional tests

peterbarker, 2016-10-24:
Please attach the telemetry log created by MAVProxy (mav.tlog).

Because you're seeing this in both MAVProxy and dronekit-python, the problem doesn't lie in dronekit-python; possibly in pymavlink, a common root for both projects.  Also possibly in Mission Planner.  Let's find out.

patrickpoirier51, 2016-10-24:
Here is the file, let me know if you require longer-shorter-other test

[mav.tlog.zip](https://github.com/dronekit/dronekit-python/files/549239/mav.tlog.zip)

peterbarker, 2016-10-24:
This is the error message that's currently being seen:

```
>>> Exception in message handler for HEARTBEAT
>>> mode 0 not available on mavlink definition
```

OK, I think @fnoop nailed this on #610.

```
pbarker@bluebottle:/tmp/fred$ mavlogdump.py mav.tlog  --t HEARTBEAT --show-source
2016-10-25 09:39:02.64: HEARTBEAT {type : 6, autopilot : 8, base_mode : 0, custom_mode : 0, system_status : 0, mavlink_version : 3} srcSystem=255 srcComponent=0
2016-10-25 09:39:02.66: HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3} srcSystem=1 srcComponent=1
2016-10-25 09:39:02.71: HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3} srcSystem=1 srcComponent=1
2016-10-25 09:39:03.31: HEARTBEAT {type : 6, autopilot : 8, base_mode : 0, custom_mode : 0, system_status : 0, mavlink_version : 3} srcSystem=255 srcComponent=190
2016-10-25 09:39:03.48: HEARTBEAT {type : 6, autopilot : 8, base_mode : 0, custom_mode : 0, system_status : 0, mavlink_version : 3} srcSystem=255 srcComponent=0
2016-10-25 09:39:03.72: HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3} srcSystem=1 srcComponent=1
```

As you can see, heartbeat packets are coming both from the vehicle (srcSystem=1) and from the ground control station (srcSystem=255).

dronekit-python sees heartbeats from every source, and fails to map the Ground Control Station's current mode number to a mode string.

This is a problem with pymavlink - it doesn't let you lock onto a specific vehicle for input.

As also mentioned on the other ticket, I do have a solution for this situation, but I haven't managed to convince tridge of the concept.  You can have a look at the "avoidance" example (https://github.com/peterbarker/dronekit-python/tree/source-system-filtering/examples/avoidance) to see how you can use dronekit-python with multiple srcSystems floating around.

Another option you might have is to connect dronekit-python to the vehicle and then connect the ground control station to dronekit-python; the avoidance example referenced allows you to do that so you can visualise the avoidance behaviour.

patrickpoirier51, 2016-10-25:
I see, but don't we have the same pattern on sitl on Ubuntu ?
Here a log file of the same test done on SITL 
[mav.tlog.zip](https://github.com/dronekit/dronekit-python/files/549366/mav.tlog.zip)

peterbarker, 2016-10-25:
Port 14550 is an output port for MAVProxy when running as SITL.  It only forwards packets from its masters there - it doesn't send its own packets there.  If you change the second MAVProxy's system id to something other than 255, then look at its mav.tlog after connecting it to SITL, you will see that the SITL's MAVProxy's system-id won't appear next to a heartbeat.

patrickpoirier51, 2016-10-25:
So in the case of connecting to Mission Planner as a simulation tool would you consider that the problem could be resolved by requesting MP to behave like SITL?

paan2097, 2016-10-25:
@peterbarker Is there proof that this error actually affects how the code runs? I get the error and is very annoying for reading print statements but at visual inspection, the vehicle seems to perform the same connected to MP or not.

patrickpoirier51, 2016-10-25:
I will check on this. My week-end project was to read joystick from mission planner, map it to channels 5-8 and send it over to my on-board Odroid XU4 companion computer so it is  controlling a 3 axis servo PTZ platform using pololu maestro.  It is working fine with SITL, I will compare with MP to see if it is as smooth (and not jerking because of the error handling). Please take note that I added --streamrate=20 on SITL so I can get full 20 Hertz data stream.

peterbarker, 2016-10-25:
On Tue, 25 Oct 2016, patrickpoirier51 wrote:

> So in the case of connecting to Mission Planner as a simulation tool would
> you consider that the problem could be resolved by requesting MP to behave
> like SITL?

Sorry, I don't think so.  MVAProxy is really playing according to its name 
at this point - proxying the output from the SITL binary onto its outputs. 
I don't think MP claims this capability - it is just adding extra "master" 
links, I think.

patrickpoirier51, 2016-10-25:
LOL , I modified this comment and removed the MP request , and apparently github sent you the unmodified message. :-).
BTW I noticed that the MP simulator calls a shell running SITL,  I will try to ''tweek'' this one to export a master at 14550. I'll keep you updated.

patrickpoirier51, 2016-10-25:
@peterbarker  looking at MP- SITL, is is using  sim_vehicle.py (2.7.10)
It shows:
Framework to start a simulated vehicle and connect it to MAVProxy.
Peter Barker, April 2016
based on sim_vehicle.sh by Andrew Tridgell, October 2011

So I presume that you know pretty much the code....

On line 57, we can read:      print("--mavproxy-args not permitted in compat mode")
Does it mean we cannot pass arguments like --out udp:my.odroid.xu.4:14550  ?

I know it may get a little off topic by it relates to the same developper

fnoop, 2016-10-25:
@patrickpoirier51 yup just use -m "--out=udp:my.odroid.xu.4:14550" and don't use -C option.

patrickpoirier51, 2016-10-25:
OK I'll make sure Scooby-Doo dont step on it ;-)

@paan2097  you are right, considering the specific setup I am working on:
 Joystick-MP(SITL)- TCP - Dronekit- Pololu - Servos ,
 I have no necessity for fast response and bidirectionnal control , so the error message is more an annoyance than a limitation. 

So Thanks you all for the support, I will keep an eye on future developments on the pymavlink option to choose a specific ID for input & control

paan2097, 2016-12-05:
@patrickpoirier51 @peterbarker 
Made it out into the field and this issue caused me problems. My drones were flying great being controlled by raspberry pi CC but stopped working when we connected to Mission Planner. I'm having problems getting the avoidance example to work because dronekit-sitl is troublesome so I usually use a MAVProxy sim_vehicle.py SITL and connect to that. Any chance this issue is being solved or should I further investigate the avoidance example?

peterbarker, 2016-12-05:
@paan2097 I'm not hopeful of getting the mav-source-system stuff into upstream pymavlink any time soon, sorry.

siliconalchemy, 2016-12-05:
@peterbarker So just to be clear, this issue is not present when using px4 now, but is still present when using arudpilot?
https://github.com/dronekit/dronekit-python/issues/610

peterbarker, 2016-12-07:
@siliconalchemy  this is really a pymavlink issue, not really related to which autopilot is being used.

LeonNie52, 2016-12-30:
For APM Planner 2,  disable GCS heartbeat transmit(for GCS fail safe ) in the config.
I think it would be the same in the Mission Planner.

tizianofiorenzani, 2017-09-08:
Hey guys, I too experienced the very same, annoying issue, even with everything at the very last release. In my case I FIXED by unistalling pymavlink and dronekit and installing back the dronekit from the github package:

```
pip unistall pymavlink (multiple times)
pip unistall dronekit (multiple times)
cd ~
git clone https://github.com/dronekit/dronekit-python.git
cd ./dronekit-python
sudo python setup.py build
sudo python setup.py install
```

I am running ArduCopter 3.4 and MissionPlanner. The problem seemed to be related with MissionPlanner, as QgroundControl worked fine.

MaEtUgR, 2017-11-30:
@tizianofiorenzani Thanks a lot for posting your solution! Fixed the issue for me as well.
first two commands you posted I had to execute only once but sudo and there's a typo: "uniNstall".
Resulted in:
```
sudo pip uninstall pymavlink
sudo pip uninstall dronekit
cd ~
git clone https://github.com/dronekit/dronekit-python.git
cd dronekit-python
sudo python setup.py build
sudo python setup.py install
```

dgorissen, 2018-03-26:
Uninstalling pymavlink/dronekit and re-installing from github per @MaEtUgR 's suggestion worked for me in that the errors still show but at least it works. So some progress since I reported the core issue 2 years (#610) ago but would be great for a proper fix.

MaEtUgR, 2018-03-26:
@dgorissen I hope I'm not offending anyone but I'm not sure about further active development of DroneKit.
Did you take a look at [DroneCore](https://github.com/dronecore/DroneCore)? Because I gave up on DroneKit after having some more issues, switched to DroneCore and was able to do all my necessary tests without any problems... It enjoys active development. But of course usability depends on your application purpose.

hamishwillee, 2018-03-26:
@dgorissen Further to @MaEtUgR comment DroneCore is optimised for PX4 autopilot - it could in theory be used with ArduPilot but has not been tested on that platform - I would not at this point recommend that approach. 

DroneKit is not getting a lot of active development but there is quite an active community doing some contribution. I'd still stick with it for ArduPilot.

dgorissen, 2018-03-27:
Thanks @MaEtUgR, @hamishwillee. I evaluated the options and given my existing codebase/approach and that I now have a workaround, I decided to stick with it.

AsternA, 2018-06-03:
Hey guys, I am also getting this error. I have read the entire thread and would appreciate some assistance.

This is the ultimate configuration I am trying to achieve:

[image]

Have the pixhawk and the GCS communicate with each other using the RPi as a tunnel, while simultaneously the RPi should be able to pull parameters from said tunnel to a GPIO pin.

My code is as follows
First terminal:
mavproxy.py --master=udpout:192.168.19.136:14550 --out=/dev/ttyS0 --out=udp:localhost:8080 --baudrate 57600

The first IP is the GCS computer IP, the pixhawk is connected through serial (RPi, /dev/ttyS0) and the second IP is the RPi localhost IP.
With this line of code, I get all telemetry working on the GCS and everything is fine.

Second terminal:
I run my script and I get the following output:
connecting on vehicle: localhost:8080

>>>Exception in message handler for HEARTBEAT
>>>mode 0 not available on mavlink definition

and nothing seems to work.

Everything was tested and was working!
Without the GCS I tested using:
mavproxy.py --master=/dev/ttyS0 --out=udp:192.168.26.78:14550 --baudrate 57600

So this places the RPi as the master while forwarding information to the RPi Wifi IP, after this, a second terminal is opened to run the code and implement a --connect “UDP:192.168.26.78:14550” and the script ran flawlessly.

From what I understand both the GCS and the pixhawk send heartbeats and so confusion happens when I try to run my script.

Any help is appreciated,

Thanks!

andrewerrington, 2018-07-24:
I still have this problem. I am running DroneKit on a Pi Zero W, linked via its on-board UART (/dev/serial0) to TELEM2 on a PixHawk running Rover v3.2.3. I followed kostkar's suggestion in #610 and I have ended up with dronekit 2.9.1 and pymavlink 2.0.6, which is what I had when I originally installed with pip yesterday.

Here's the error I see:

Exception in message handler for HEARTBEAT
mode 1 not available on mavlink definition

I had it intermittently yesterday, and I could make it appear when connecting onto TELEM1 with MissionPlanner, and disappear when I disconnected. Now it's there most of the time, and I am getting timeouts due to lost heartbeats.

I tried the hellodrone.py example from the 3DR dronekit website. Today I am just trying this:
'''
Import DroneKit-Python

print("Importing DroneKit.")
from dronekit import connect, VehicleMode

Connect to the Vehicle.

print("Connecting to vehicle.")
vehicle = connect('/dev/serial0', baud=115200, wait_ready=True)

Get some vehicle attributes (state)

print "Get some vehicle attribute values:"
print " GPS: %s" % vehicle.gps_0
print " Battery: %s" % vehicle.battery
print " Last Heartbeat: %s" % vehicle.last_heartbeat
print " Is Armable?: %s" % vehicle.is_armable
print " System status: %s" % vehicle.system_status.state
print " Mode: %s" % vehicle.mode.name # settable
'''
Am I missing something? This thread seems to have gone quiet for a while.

vishchiv, 2018-11-19:
I ran into this issue while connecting my program and Mission Planer to SITL.
FIXED by setting lower port for program(5763 >> 5762) and higher for Mission Planer(5762 >> 5763)

blakethompson82, 2018-11-26:
How did you set ports for both the program and Mission Planner? I am new to this and I have already updated dronekit, pymavlink, and everything else mentioned in this thread with no luck.

hamishwillee, 2018-11-26:
@blakethompson82 Read the guide http://python.dronekit.io/develop/sitl_setup.html - at the end it shows how to set an additional port for a GCS

maciek01, 2019-11-17:
so what's the fix ? i just connected my code (which works great with matek f765 wing 4.0.0rc1) to CUbe Orange (4.0.0rc2)
and cant past this error.

this is raspi talking directly to FC via /dev/serial0

here is my config:

sudo apt-get install git python-pip python-dev screen python-wxgtk2.8 python-matplotlib python-opencv python-numpy libxml2-dev libxslt-dev ppp elinks python3-lxml python-lxml dnsutils cu sqlite3 python-pysqlite2 -y

sudo pip install --upgrade future
sudo pip install --upgrade httplib2
sudo pip install --upgrade pyserial
sudo pip2 install --upgrade dronekit

tizianofiorenzani, 2019-11-19:
Build and install dronekit from source

maciek01, 2019-11-19:
a bit excessive for a python library :)

Th3OnlyN00b, 2020-02-27:
Is this still required, building from source?

maciek01, 2020-02-27:
> Is this still required, building from source?

for me it is the only way to get it to installed successfully if i want to stay on the latest dependencies.
i have no time or patience to experiment - so for me the full build is the path of least resistance

Th3OnlyN00b, 2020-02-27:
Thanks for the fast reply!

bduva002, 2020-04-24:
@maciek01 i am getting the same error as you. I built from source following this documentation. 

https://dronekit-python.readthedocs.io/en/latest/contributing/developer_setup_linux.html 
Is this what you did to fix this error?

ERROR:dronekit:Exception in message handler for HEARTBEAT
Traceback (most recent call last):
  File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 1527, in notify_message_listeners
    fn(self, name, msg)
  File "build/bdist.linux-armv7l/egg/dronekit/__init__.py", line 1219, in listener
    raise APIException("mode (%s, %s) not available on mavlink definition" % (m.custom_mode, m.base_mode))
APIException: mode (0, 4) not available on mavlink definition

Th3OnlyN00b, 2020-04-24:
Yes, building from source worked for me.

bduva002, 2020-04-24:
Okay so I found were the error is coming from in the code. It when the code checks to see if a mode is available. 
here is the line:
https://github.com/dronekit/dronekit-python/blob/fe6c62313831c7d4847ae8009af3f6400aea8c5e/dronekit/__init__.py#L1218

what I don't understand is when we go to the is_mode_available def.  If I am running arduplane wont this function always return false and the mode not available message keep coming back?  

[image]

I am using Arduplane 4.0.X, dronekit used to work before on older version of arduplane.

bduva002, 2020-04-24:
@Th3OnlyN00b thanks for a reply.  What version of ardupilot are you using? 
Is it 4..0.0 or newer?
Are you using a cube orange?

> Yes, building from source worked for me.

When you built from source did you follow the the instructions in this link?
https://dronekit-python.readthedocs.io/en/latest/contributing/developer_setup_linux.html

zlowell2, 2021-05-05:
Has there happen to be any update on this thread? I am currently building dronekit from source and connecting to a cude orange (running ardupilot 4.0.7) to raspi4 communicating via serial to telem2 port and am getting the same error as @bduva002 above. Any help would be great.

CodingGhost, 2022-04-29:
Im in the same boat

xianglic, 2025-01-15:
what does mode(custom_mode, base_mode) represent? if it is mode(0, 4) then, how do we find the mapping to retrieve its semantic meaning? and how does the added ADBS board relates to these two modes?