dronekit/dronekit-python #682: can't update waypoints from file

elgeografo, 2016-11-15:
Hi, I'm triying to upload some waypoints from a file to pixhawk. everithing is ok util i try to update the commads, the program does not respond. Here is a code snipped.

cmds = vehicle.commands
        cmds.clear()
        #Add new mission to vehicle
        for command in missionlist:
            cmds.add(command)
        print ' Upload mission'
        cmds.upload()
        print ' Mission uploaded'

The program never print ' Mission uploaded'. In debug mode the program stops in cmds.upload()

I have reviewed the file strcuture and everething it seems ok. Any help would be very welcome.

Thanks,

peterbarker, 2016-11-15:
On Tue, 15 Nov 2016, elgeografo wrote:

> The program never print ' Mission uploaded'. In debug mode the program stops
> in cmds.upload()

Please try a simple parameter set and see if that sticks.

hamishwillee, 2016-11-15:
Also tell me what you see out of http://python.dronekit.io/examples/mission_basic.html and http://python.dronekit.io/examples/mission_import_export.html

These examples are known to work when last tested.

elgeografo, 2016-11-16:
Ok, I think I see the problem. The vehicle needs to be in "is_armable" state ¿right?. If not is_armed the operation can not be performed. I'm using a pixhawk indoor, so it does not have 3DFix GPS signal. May I skip this state in order to upload the mission?. Thanks

hamishwillee, 2016-11-16:
I don't know, but I don't think so. My gut feeling is that you should be able to upload a mission before it gets to this state BUT missions do require a GPS fix, so I suspect that is your problem.

peterbarker, 2016-11-17:
On Wed, 16 Nov 2016, Hamish Willee wrote:

> I don't know, but I don't think so. My gut feeling is that you should be
> able to upload a mission before it gets to this state BUT missions do
> require a GPS fix, so I suspect that is your problem.

This is the "wait_ready" stuff biting people.

You *can* pass "wait_ready=False" in (or somesuch....) to the connect 
IIRC.  You just have to be sure to do something equivalent *after* 
uploading the mission...

elgeografo, 2016-11-18:
Peter,
In fact if I start the connection via radio with "wait_ready=True" I do not receive any message, I have to pass it to False (If I connect my pixhawk via usb, I receive messages anyway, true/false).

Thank you all very much for your help

closed 2016-11-18 by elgeografo