dronekit/dronekit-python #684: Error running vehicle_state.py sample against simulator: WPM: REJ. CMD: Req.  WP was unexpected

duboisj, 2016-11-22:
I am running a simulator in a docker image I produced by following instructions in the PX4 Dev Guide.  The vehicle_state.py example from dronekit-python appears to fail at the cmds.wait_ready() step on line 112.  The failure is:

>>> Home position: 47.3977419, 8.5455934, 488.10
>>> WPM: REJ. CMD: Req. WP was unexpected

After that, the wait_ready call times out.

I notice that the script has output the Firmware version which it thinks it has connected to as this:
Get all vehicle attribute values:
 Autopilot Firmware version: PX4Copter-1.5.0-dev0
   Major version number: 1
   Minor version number: 5
   Patch version number: 0
   Release type: dev
   Release version: 0
   Stable release?: False

That looks quite different (in terms of version) from what I see on the current dronekit-python site here: http://python.dronekit.io/examples/vehicle_state.html .

Is the docker image I am running from hopelessly out of date?  I just downloaded and created it from what appeared to be current instructions from the PX4 Dev Guide ... but I had a problem with pymavlink also.  

Should I not rely on that image, and instead try to install or build everything myself?  Or does the PX4 Dev Guide need to be updated so the instructions give a more up-to-date image?  Or have I likely done something else wrong entirely?

peterbarker, 2016-11-22:
On Mon, 21 Nov 2016, duboisj wrote:

> Autopilot Firmware version: PX4Copter-1.5.0-dev0
> That looks quite different (in terms of version) from what I see on the
> current dronekit-python site here:
> http://python.dronekit.io/examples/vehicle_state.html .

The example output is from ArduPilot, not PX4Copter.

> Is the docker image I am running from hopelessly out of date? I just

Possibly - but that's not the source of your confusion :-)

If you want to play with dronekit-python then you can follow the 
quick-start guide (which uses ArduPilot, not PX4 Pro: 
http://python.dronekit.io/guide/quick_start.html

As you've discovered, dronekit-python's support for PX4 Firmware is 
not complete.

duboisj, 2016-11-24:
Oops!  Thank you.  All things drone are new to me and I'm still trying to figure out how the pieces fit together.  My hardware is 3DR (Solo) and it looked initially like dronekit-python would be a good choice (both for the simulator and talking to real hardware).  Obviously I'm still getting oriented, though :-)

mymultiverse, 2018-12-18:
Is the dronekit-python's support for PX4 Firmware still not complete?

peterbarker, 2018-12-18:
On Mon, 17 Dec 2018, Nebula wrote:

> Is the dronekit-python's support for PX4 Firmware still not complete?

Correct.  It's support for ArduPilot less so - but likewise!

Note that some recent updates to pymavlink might help things along a bit. 
Missions are still likely to be a problem, 'though.