dronekit/dronekit-python #685: Autonomous Indoor Flight with Computer Vision (Channel Override?)

GitWyd, 2016-11-25:
We read that we should consult you first to figure out if there is a better solution than Channel Overrides for our problem.

We are a research team working on autonomous UAV flight in GPS deprived environments.

In order to do that, we need to

1. Arm the drone without GPS (is the "GUIDED_NOGPS" mode the right one for this?)
2. Send receiver commands to the drone: throttle, roll, pitch, yaw (Preferably with a magnitude) or other commands that allow us to directly control the drone i.e. change direction, ascend/descend, forwards/backwards

Eventually we will want to program aggressive flight maneuvers, but initially we simply need to have basic controls.

Does DroneKit provide some of those functions already? Which functions should we consider? Which functions do we need to code ourselves?

We are happy to contribute to the DroneKit project, if this is necessary to achieve the above described functionality.

Our Setup:
Navio2 with PI3 on custom build 330 size quadcopter

peterbarker, 2016-11-25:
On Fri, 25 Nov 2016, GitWyd wrote:

>  1. Arm the drone without GPS (is the "GUIDED_NOGPS" mode the right one for
>     this?)

Probably.

>  2. Send receiver commands to the drone: throttle, roll, pitch, yaw
>     (Preferably with a magnitude) or other commands that allow us to
>     directly control the drone i.e. change direction, ascend/descend,
>     forwards/backwards

You probably don't want to do this; you probably want to send guided 
velocity mavlink messages instead: 
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L2962

> Eventually we will want to program aggressive flight maneuvers, but
> initially we simply need to have basic controls.
> 
> Does DroneKit provide some of those functions already? Which functions
> should we consider? Which functions do we need to code ourselves?

People have used dronekit to push their vehicles around using the guided 
velocity commands; 3DR's Solo vehicle's smartshots, for example.

squilter, 2016-11-26:
`Guided-NoGPS` only allows you to send angle commands. You can send `SET_ATTITUDE_TARGET` messages, and you can listen to `ATTITUDE` messages.

If you attach an optical flow module and a rangefinder, then you can use normal guided mode. This will allow you to send velocity commands or even position commands.

markrjr, 2017-02-22:
I have been doing some testing and the `SET_ATTITUDE_TARGET` messages do not work, and there's no way to set `Guided-NoGPS`.

dollop80, 2017-04-25:
@squilter 

> If you attach an optical flow module and a rangefinder, then you can use normal guided mode. This will allow you to send velocity commands or even position commands.

I've added the OF and lidar. It works indoors without GPS **only** when arming in Loiter mode. No way to use OF in Guided mode. I've tried AC_3.4.5 and AC_3.5RC3

studroid, 2017-05-07:
@dollop80 
Refer to #712 that I wrote.
This is probably what you are looking for!

prasanna12510, 2017-11-25:
is there any simple code like arm and takeoff that demonstrates guided_nogps flight mode using dronekit api

hamishwillee, 2017-11-26:
@prasanna12510 Yes. See https://github.com/dronekit/dronekit-python/pull/712 - not merged, but some great work there.

borhanreo, 2019-07-15:
> @squilter
>
>
> I've added the OF and lidar. It works indoors without GPS **only** when arming in Loiter mode. No way to use OF in Guided mode. I've tried AC_3.4.5 and AC_3.5RC3

@dollop80 I have used maxbotix rangefinder sensor. So may i arm and takeoff in LOITER mode in indoor (GPS denied area).

rajivrawat982, 2020-09-17:
I'm just looking to find out how flight controller deciding the actual altitude in "GUIDED_NOGPS" mode as while i was arming it without a props and using [vehicle.location.global_relative_frame.alt] to find out current altitude it changing continuously, it should not change as my drone is not moving only arming.