dronekit/dronekit-python #687: ImportError : cannot import name connect

Appassionato, 2016-12-16:
The title says the problem. I'm trying to use dronekit on a Raspberry Pi. I have 2 of them . I tried writing a simple code to connect the vehicle but I get the error " cannot import name connect " . Then I went on my older Pi where I had a dronekit program stored there and tried it there. Only managed to run it once. Every subsequent times I get the same error. Anyone has a fix ? Thanks!

hamishwillee, 2016-12-18:
No idea. Have you run the vehicle_state.py example in the docs?

MKSRIRAM1993, 2017-11-15:
Hello Appassinoato... I too have same problem. when i run the scrip. it shows "cannot import name connect", but when i type the same in the python command line its working... did you fix error? if you have pls share.....

MKSRIRAM1993, 2017-11-16:
hello hamishwillee..... 
i have installed only 2.7 version

hamishwillee, 2017-11-16:
@MKSRIRAM1993 No, it is EXTREMELY likely you have two versions installed - probably both 2.7
Usually this bug is because the pip that you are using to install dk is different than the Pip used by the run environment. 
You can verify this by doing `python -v` in the command line and writing a script that outputs the same information. If they are different then you have two versions. 

If that isn't the problem, then I have no idea.

MKSRIRAM1993, 2017-11-16:
thank you for the quick reply........I tried running a python script with only 
python -v
(sry i am noob to python)
i got a error " name "python" is not defined" 
so if it is different Pip, then how should i proceed...

hamishwillee, 2017-11-16:
I updated my comment above to do it step by step. Google search is your friend.

MKSRIRAM1993, 2017-11-16:
thank you hamishwillee....

soufyan788, 2020-05-30:
I get this problem when i tried to run my code to connect to my drone
ImportError: cannot import name VihecleMode

can anyone help me,please