dronekit/dronekit-python #689: Channel Override Alternative

jodumagpi, 2016-12-27:
I have been using channel override to specify the approximate direction of the drone (angle rotation) and control the speed as it moves forward. This is so that the drone can land precisely on a target based on it's approximate horizontal and vertical distance. However, I've read on the documentation that this is "highly dis-commended". If so, what are the alternative commands?

peterbarker, 2016-12-27:
On Mon, 26 Dec 2016, Jo Dumagpi wrote:

> I have been using channel override to specify the approximate direction of
> the drone (angle rotation) and control the speed as it moves forward. This
> is so that the drone can land precisely on a [...]

guided mode, specifying angles at which the aircraft should be oriented.

cheintz, 2017-03-20:
Is there a similar function for fixed wing? Preferably one that behaves similarly to FBWB? It seems guided mode for fixed wing sets target roll and pitch angles, rather than any direct transnational velocities.

studroid, 2017-05-07:
You may refer to this: #712 
Changing attribute is more likely what you are looking for.

closed 2019-10-28 by jodumagpi