dronekit/dronekit-python #690: PX4 mode bug

sanderux, 2016-12-27:
@mrpollo after this PR has merged, could you bump the pymavlink version to re-enable support for PX4?
https://github.com/ArduPilot/pymavlink/pull/42

/cc @LorenzMeier

mrpollo, 2016-12-27:
Sure thing, just ping me when it gets merged

sanderux, 2016-12-28:
@mrpollo ping

mrpollo, 2016-12-28:
hey @sanderux pymavlink needs a new release, it looks like it hasn't been released since [2.0.8](https://pypi.python.org/pypi/pymavlink/2.0.8) (thats the latest i could find on pypi), we need them to create a release and push that to pypi before I can make a new release

sanderux, 2017-04-20:
@mrpollo @peterbarker at my request tridge was kind enough to release pymavlink 2.2.0 on pypi. could you bump the version requirement in dronekit-python?

peterbarker, 2017-04-21:
On Thu, 20 Apr 2017, Sander Smeets wrote:

> @mrpollo @peterbarker at my request tridge was kind enough to release
> pymavlink 2.2.0 on pypi. could you bump the version requirement in
> dronekit-python?

I've changed it from a strict 2.0.6 requirement to >= 2.0.6.

I suspect PX4 Flight Core requires >= 2.2.2, 'though?

sanderux, 2017-04-21:
It does

sanderux, 2017-07-07:
Seems to be solved, thanks guys

closed 2017-07-07 by sanderux