dronekit/dronekit-python #694: dependency problem in installation

credtiger96, 2017-02-02:
I tried to install dronekit and found some problem. 

``` 
pip install dronekit 
``` 
does not work anymore and I also found that 
pymavlink==2.0.6 makes problem but 
it seems pymavlink==2.0.6  cannot be installed via pip.

lroberts99, 2017-02-03:
I have this issue as well. Is there a resolution to this problem? 

Running ` pip install dronekit` prints this error:

` Could not find a version that satisfies the requirement pymavlink==2.0.6 (from dronekit>=2.0.0b6->dronekit-sitl) (from versions: 2.0.8, 2.1.0)
No matching distribution found for pymavlink==2.0.6 (from dronekit>=2.0.0b6->dronekit-sitl)
`

dagar, 2017-02-03:
If you change the requirements.txt to 2.0.8 does it work?

peterbarker, 2017-02-06:
tridge has re-uploaded 2.0.6.  It's actually 2.0.8 in disguise.  That might cause people issues.  Let's find out.

We're going to do another release of dronekit-python with less restrictive pymavlink version requirements...

Please retry and reopen this issue if its still broken (or differently broken).

closed 2017-02-06 by peterbarker

Puru-1236, 2021-04-22:
I am also getting an issue while setting up dronekit .Error is:
ERROR: Could not find a version that satisfies the requirement dronekit (from versions: none)
ERROR: No matching distribution found for dronekit