dronekit/dronekit-python #695: Copter not arming through python script

Biswajit1995, 2017-02-11:
Hello all,

I'm trying to make my drone fly without GPS and to move to certain distance by overriding RC commands, for maintaining position I am using optical flow camera and a lidar, but the very first thing is I'm not able to arm my copter through my script.

I'm using this following code:

```
#/usr/bin/env python
# -*- coding: utf-8 -*-
from dronekit import connect,VehicleMode,LocationGlobalRelative
from pymavlink import mavutil
import time
# Receiving from command line
import argparse
parser = argparse.ArgumentParser()
parser.add_argument('--connect',help="PORT_NO")
args = parser.parse_args()

# Connecting to the vehicle
connection_string = args.connect
print("Connecting to...% s" % connection_string)
vehicle = connect(connection_string, wait_ready=True)

#Function to arm and takeoff to a specified altitude
def arm_and_takeoff(aTargetAlt):
	print("Basic Prearm checks..dont touch!!")
	print("Waiting for vehicle to initialize")
	time.sleep(1)
	print("Arming Motors..")
	# Copter should arm in Guided-mode
	vehicle.mode = VehicleMode("STABILIZE")
	
	while not vehicle.is_armable:
		print("Waiting for vehicle to initialize...")
		time.sleep(2)

	vehicle.armed = True
	vehicle.flush()
        print("Is Armed:% s"% vehicle.armed)
	
	while not vehicle.armed:
		print("Waiting for arming...Mode:% s"% vehicle.mode)
		print("Is Armed:% s"% vehicle.armed)
		vehicle.armed = True
		time.sleep(2)
		
	print("Taking Off..")
	vehicle.simple_takeOff(aTargetAlt)

	while True:
		print("Altitude: ",vehicle.location.global_relative_frame.alt)
		#Break and return from function just below target altitude.
		if vehicle.location.global_relative_frame.alt>=aTargetAlt: 
			print("Reached Target Altitude..")
			break
		time.sleep(2)
	
vehicle.close()
arm_and_takeoff(50)
```

But its printing:

```
Basic Prearm checks..dont touch!!
Waiting for vehicle to initialize
Arming motors..
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
Waiting for vehicle to initialize....
```
And it keep printing like this, i don't know what I'm missing or doing wrong.
And to note i have` no GPS connected, only PX4Flow and Lidar`, that's why I'm trying to takeoff in STABILIZE mode.

Any help would be greatfull.

Thanking you
Biswajit

srex212, 2017-02-23:
Hi Biswajit,

The issue is with the code:

while not vehicle.is_armable:
		print("Waiting for vehicle to initialize...")
		time.sleep(2)

I, personally, have found that the vehicle will arm even is vehicle.is_armable is false. I have tried to adjust every setting just to see if vehicle.is_armable will become True. I couldn't find anything that worked for me. Just remove that segment of code. It will probably arm (at least it did for me, although I had GPS...)

Biswajit1995, 2017-03-01:
Hey srex212,

Thanks for the solution, and im too glad that my code is working.
By the mean time i was working with RC Overrides to control individual
channels, by providing values and it worked. But my motive was to create an
mobile application that will act as the transmitter, and Im not sure how to
do that.

Like, the mobile screen will be having two joystics to replicate as the
transmitter, and i could do that, but how to send those values to raspberry
pi and the next is how to use that values inside my scripts.. If you can
give me any advice regarding this I will be grateful.

Thanks and regards
Biswajit

AhmedRejeb, 2017-07-03:
hello @Biswajit1995 i'm working in the same project and did try your code but it didn't work for me.
it keeps telling me 
waiting for arming .. Mode:VehicleMode:STABILISE 

what should i do 
@srex212  can you help me with his essue .
this how i made the connection :
i did run this commande in a bash
`sudo mavproxy.py --master=/dev/serial0 --out=udpout:127.0.0.1:14550`

and in the same time i did run 
`sudo python TakeOff2.py --connect 127.0.0.1:14550`
in the other Bash 
[image]

AhmedRejeb, 2017-07-03:
i want to add this too
when i want to arm it the mavproxy like this .it work but when takeoff i have this error 
[image]

bhaskar-glitch, 2021-09-17:
> Hey srex212, Thanks for the solution, and im too glad that my code is working. By the mean time i was working with RC Overrides to control individual channels, by providing values and it worked. But [...]

Hey,Please can you tell me that this RC OVERRIDIN...is it working without gps??
if yes then can you please help me out with the code...
i'll be thankfull to you