dronekit/dronekit-python #696: Failing to connect to drone from laptop over network

deb0ch, 2017-02-12:
Is connection from remote computer over network currently supported ?

I have an ErleCopter 3, to which I'm connecting from my laptop either over wifi or network cable. I want to remotely control my drone from my laptop using DroneKit and a network connection.

I'm running a simple program which just tries to connect to the drone:

``` python
from dronekit import connect

copter = connect('udp:10.0.0.1:6001', wait_ready=True)
```

Regardless the ip and the port (I've made sure that I used the right ones, tried with both tcp and udp), as long as it is a remote ip address, it fails with the exception:

```
Traceback (most recent call last):
  File "dk_gimbal_control.py", line 5, in <module>
    copter = connect('udp:10.0.0.1:6001', wait_ready=True)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/__init__.py", line 2817, in connect
    handler = MAVConnection(ip, baud=baud, source_system=source_system, use_native=use_native)
  File "/usr/local/lib/python2.7/dist-packages/dronekit/mavlink.py", line 128, in __init__
    self.master = mavutil.mavlink_connection(ip, baud=baud, source_system=source_system)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 1214, in mavlink_connection
    return mavudp(device[4:], input=input, source_system=source_system, use_native=use_native)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 875, in __init__
    self.port.bind((a[0], int(a[1])))
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address
```

which is odd, because as far as my understanding goes I thought that DroneKit was acting as a client of the Ardupilot instance instance in the drone, and AFAIU `bind` should only be called on the server side, to open the requested port ?

If DroneKit is supposed to act as a server, how does APM initiate the connection then ? Or does it ?

Is remote connection over network simply not supported, or did I miss a step / port / whatever ?

rtfswr, 2017-02-12:
think of DroneKit as a GCS then everything becomes clear, i.e. you have to
send the MAVLINK messages to the system running Dronekit and then initiate
the connect.

deb0ch, 2017-02-12:
Precision: it is ArduCopter that runs at the address/port on which I'm trying to connect.

I thought that Dronekit was meant to send MavLink commands to Ardupilot in order to control the drone ?

I precisely thought of DroneKit as a GCS and in my understanding the GCS connects directly to a Ardupilot to exchange MAVLink packets ?

We could make it work somehow by running MavProxy on the drone, but it feels a little bit hacky (and sometimes doesn't work for reasons that I can't explain).

I don't understand exactly what you mean by

> you have to send the MAVLINK messages to the system running Dronekit and then initiate the connect.

From what to what am I supposed to send Mavlink messages before calling `connect` ?

rtfswr, 2017-02-13:
S ErleCopter 3 is sending mavlink messages on port 6001? If yes, then connect your notebook over wifi to the ErleCopter 3 first and then try to connect on udp:0.0.0.0:6001. 

We use here the MAVProxy bridge for the connections over different networks, see here http://ardupilot.github.io/MAVProxy/html/getting_started/quickstart.html. DroneKit should work, if you are on the same network.

ARPIT355, 2017-08-31:
Can you list total hardware requirements to built this drone.I will be thankful if you explain it to me.

AriouatI, 2018-02-26:
Hello,
did you succeed ? i'm trying to do pretty much the same thing: connect a Bebop2 to my computer it's allready connected through wifi but it's not working even with _rtfswr_'s connectionString.
any help.
thx

deb0ch, 2018-02-26:
In the end it's very strange and like a reversed connection.

To connect a script on my computer to the drone I had to provide the script with my own computer's IP and have the drone initiate the connection to my computer.

Yes, this is messed up.

AriouatI, 2018-02-26:
if i understand well:
1. you connected the script to the computer using 
`copter= connect('MY_IP', wait_ready=True)`

2. then use MAVProxy to connect the drone to the computer.

and then use dronekit on object "Copter".

correct ?

AriouatI, 2018-04-23:
hello again,
after trying to work with different other modules i came back to dronekit but i can't get exactly what you did.
i've tried establishing a connection using this line
` mavproxy.py --master tcp:127.0.0.1:5760 --out udp:127.0.0.1:14551 --out udp:192.168.42.3:14550`

where 192.168.42.3 is the drone's ip
as i was connected to the bebop's wifi but the connaction fails.
could you guide me a little bit more.
Thanks a lot for your precious and priceless help.

deb0ch, 2018-11-06:
¯\_(ツ)_/¯

MuhammadFaizanKhan, 2018-11-06:
?? Don't know or didn't understand my problem:)

MuhammadFaizanKhan, 2018-11-12:
Disappointing that no success story i found yet. If future reader are reading this, then please share that how did you connect your drone to a remote server where drone kit is running. 

Dear @deb0ch whats your output about MavROS?

hamishwillee, 2018-11-12:
@mrpollo Can you comment about running dronekit as a cloud service and connecting to multiple devices? See https://github.com/dronekit/dronekit-python/issues/696#issuecomment-437875868 above.

mrpollo, 2018-11-13:
`dronekit-python` supports tcp/udp through `pymavlink` (example: [mavutil.py#mavtcp](https://github.com/ArduPilot/pymavlink/blob/master/mavutil.py#L979)), as long as the native python class `socket` can resolve the `AF_INET` tcp/udp address it doesn't matter if this is 1cm next to you or across the internet.

Having said that, making your drone available online requires extra configuration that should be covered elsewhere, it might require setting up a VPN and calling back "home" to notify of your availability, we don't have a "standard" way of doing this, its up to each developer to figure out this part.

I would also like to point out that I haven't heard of anyone running with this configuration and I'm unsure of how robust `mavutil` can get under unreliable tcp connections, I strongly recommend stress testing tcp connection across the internet before attempting any remote operation of an aircraft that might endanger you or anyone nearby.

mrpollo, 2018-11-13:
I would also like to recommend [`mavlink-router`](https://github.com/mavlink/mavlink-router) a blazing fast native mavlink routing alternative to [`MAVProxy.py`](https://github.com/ArduPilot/MAVProxy) the developer control station, both great projects, choose whatever fits your workflow the best.

hamishwillee, 2018-11-13:
I doubt there is a more general solution - your client is going to have to phone home, because its network/IP might change and the server can't know it. This is why clients connect to fixed servers, and not the other way round.

hamishwillee, 2018-11-13:
I am closing this. The solution is not in DroneKit itself.

closed 2018-11-13 by hamishwillee

deb0ch, 2018-11-14:
@hamishwillee Hi, I didn't understand your last statement. Can you elaborate on why the connection is reverse ? What do you mean by `This is why clients connect to fixed servers, and not the other way round.` ? is what you call `client` the drone, or the GCS ? And `server` ?

What does `your client is going to have to phone home` mean ? Whose IP might change, the one of the drone or of the GCS, under what circumstances ?

hamishwillee, 2018-11-15:
@deb0ch 
Normally when you use dronekit or mavlink you connect to a device on a fixed IP for that drone from the same network. 
However in the website scenario you don't know the IP of the drone because it can change and because you aren't necessarily connecting to a specific drone, but rather to some drone that wants to use your service. 
You only know the website because its IP address (or at least its DNS address) can be stable. So your vehicle has to somehow "phone home" to the website to start the connection, rather than what we normally do. 

The analogy of client and server is really just to explain what needs to initiate connection. You could look at this just like github. Github is capable of sending messages to your browser when you are connected, but it can't do so until you have logged in, because until you do so, it does not know your IP. 

HOpe that makes sense.

deb0ch, 2018-11-15:
Ok, I see how this makes sense for some use cases, like the website use case. But it doesn't work for many others, like the guy who only owns one drone and tries to control it from his computer over a local network.

There are several ways of repeatably identifying a drone on a network. One of them is (like we do) attribute them fixed IPs. Another way is to use a service discovery software, like Bonjour or Avahi, which we ended up using too, that dynamically resolve a host name to an IP on a local network. In these scenarios you can even have your GCS being the only one to not have a fixed IP on the network (as it was our case).

The way it is working right now obligates the user to ssh to the drone and initiate the connection from there. You can't just turn on your drone and connect to it.

It's only my uninterested two grains of salt, as we are not using Dronekit anymore anyways. We switched to Mavros for a few reasons including this one. But I think it is holding the project back for getting users.

hamishwillee, 2018-11-16:
@deb0ch Yes, we're only talking about the website use case, because that was the question that was asked. The point to the user was that the website model can only work if the drone is able to initiate the conversation (and yes, using avahi or similar makes sense). But that is outside the scope of the current DK-Python release.
In other words, we're in agreement :-)

jinkalr-itpath, 2020-10-01:
I want to connect DJI drone, can I use droneKit API? its urgent please

naghmehivaki, 2021-09-30:
Did anyone solve this issue? I need help on this.