dronekit/dronekit-python #698: ATT_POS_MOCAP Message

andersonn, 2017-02-21:
Hi everyone,

We're trying to send the att_pos_mocap message to a drone with a pixhawk rpi combination. We have from a motion capture system the xyz and attitude values of the drone. We've tried using the message_factory.att_pos_mocap.encode to send these messages but it doesn't seem to be working. Pixhawk seems to support this mavlink message. Any tips on how to get it to work?

lalal321, 2017-09-18:
Hello , How are you doing now with the att_pos_mocap , I am doing this too. Have you solving it ？

pietrodn, 2018-02-18:
The problem you have is probably related to timestamps.
If the `ATT_POS_MOCAP` message has a timestamp "too old" relative to the autopilot's time, [the PX4 firmware detects a timeout](https://github.com/PX4/Firmware/blob/b5be990109a00814102005b6b809d64daadc40d8/src/modules/local_position_estimator/sensors/mocap.cpp#L117). If a timeout is detected, [the message is ignored](https://github.com/PX4/Firmware/blob/b5be990109a00814102005b6b809d64daadc40d8/src/modules/local_position_estimator/sensors/mocap.cpp#L59).

So, you *must* sync the Pixhawk's clock with the RPI's clock (e.g., by using the [MAVLink TIMESYNC messages](https://groups.google.com/forum/#!topic/mavlink/vz3msJllpSU)) and make sure that the timestamps in the ATT_POS_MOCAP messages are "not too old" w.r.t. that clock.