dronekit/dronekit-python #700: Sometime, Dronekit 2.9.0 does not work uploading new waypoint when FC is in armable.

kevinjhur, 2017-03-06:
Hi,

Currently I've implemented mission flight app with Dronekit-python.
My environment is as below.

FC : Pixhawk + Arducopter firmware(latest version)
MC : Raspberry Pi + Ubuntu Mate 16.04
Dronekit version : 2.9.0

When FC is not on armable just like indoor, upload command works fine.
But, when FC is on armable, sometime upload command does not return.
After then FC go to lockup, no heartbit and UART port dead.

BTW, when I try uploading new waypoint via Mission planner or Qgroundcontrol, it work fine.

Please let me know what is root cause of this issue.

tribbloid, 2017-03-20:
does not return? you mean hang up in function call or timeout after x seconds?
I may encounter the same issue before

kevinjhur, 2017-03-31:
Yes right, vehicle.commands.upload() never return. And my code is hang up.
Do you have any solution?