dronekit/dronekit-python #701: Allowing an RC controller to override

sangmino, 2017-03-18:
Are there any dronekit commands that allow an RC controller to override and take over commands sent by the app? For safety I want to be able to have the RC control the drone for backup incase the app malfunction or crashes mid flight.

studroid, 2017-05-07:
The example code named 'channel_override' seems to help you!

hamishwillee, 2017-05-07:
When you're in the mode used for control via dronekit the RC controller disables the sticks. However you can still change to other modes like auto or manual. So there is no programming to do - in emergency, just change back to a mode that uses RC.

The channel override basically allows you to send whatever you like to the output channels (ignoring the input). So yes, you could just pass through whatever is coming on the input to the output. That is a bit pointless though, since you can just switch to a manual mode.

closed 2017-05-07 by hamishwillee