dronekit/dronekit-python #704: Dronekit fails to connect to serial port

larsks, 2017-03-31:
I am using dronekit 2.9.0.  I am unable to get dronekit-python to connect to a serial port on two different systems (Linux and Mac OS) on which mavproxy connects without a problem.  That is, I can run:

```$ mavproxy.py --master /dev/ttyUSB0 ```

And get:

```Connect /dev/ttyUSB0 source_system=255
Failed to load module: No module named adsb. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from /dev/ttyUSB0
fence breach
Pkzzonline system 1
POSHOLD> Mode POSHOLD
APM: APM:Copter V3.4.6 (e707341b)
APM: PX4: de6b667d NuttX: 8c965992
```

But if I try to connect with dronekit:

```$ python2
Python 2.7.13 (default, Dec 26 2016, 20:43:09) 
[GCC 6.2.1 20160830] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import dronekit
>>> v = dronekit.connect('/dev/ttyUSB0')
```

I get:

```>>> Link timeout, no heartbeat in last 5 seconds
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python2.7/site-packages/dronekit/__init__.py", line 2827, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/home/pi/.local/lib/python2.7/site-packages/dronekit/__init__.py", line 2100, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.
```

alduxvm, 2017-06-14:
I had the same issue... 

I figure it out you need to specify the baud rate... my connection string was:
`vehicle = connect('/dev/serial0', wait_ready=True)`

and I change it to (specifying the baud rate):
`vehicle = connect('/dev/serial0', wait_ready=True, baud=57600)`

and now it works!!

illu92, 2017-12-15:
I have the same issue as larsks.

I tried to put like alduxvm says, but nothing works. Trying also with /dev/ttyUSB0

It works for me with WiFi connection, but i prefer to work with my telemetry modem

pietrodn, 2018-02-18:
I think this issue may be closed, since a working example is now included in the docs.

hamishwillee, 2018-02-19:
I agree. CLosing

closed 2018-02-19 by hamishwillee

gbaweja, 2018-10-18:
Link timeout, no heartbeat in last 5 seconds
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "wayPoint1.py", line 24, in <module>
    vehicle = connect(args.connect, baud=57600, wait_ready=True)
  File "build/bdist.linux-x86_64/egg/dronekit/__init__.py", line 3018, in connect
  File "build/bdist.linux-x86_64/egg/dronekit/__init__.py", line 2262, in initialize
dronekit.APIException: Timeout in initializing connection.

This is what i am getting when i am running..
python waypoint1.py --connect "/dev/ttyUSB0"

My pixhawk is connected with Intel UpBoard on the USB port. I am using FTDI cable.
Please suggest what am i doing wrong??

dronekit              2.9.1  
pymavlink           2.2.20  
dronekit-sitl         3.2.0