dronekit/dronekit-python #713: What does channel override numbers mean?

toashwin, 2017-05-03:
Hi 
I was trying to understand the code for channel overrides. From the following line, what does the numbers means - 110, 210, 310? 

vehicle.channels.overrides = {'1': 110, '2': 210,'3': 310,'4':4100, '5':510,'6':610,'7':710,'8':810

I understand Ch1 =Roll, Ch 2=Pitch,  Ch 3=Throttle,  Ch 4=Yaw
Am I correct?

How do I control the Speed and time? For example I want lift for 1sec. and stop. I want to go very slow.
Can I lift off, for 0.1sec at the slowest speed possible?

May be you could document this on the link:
http://python.dronekit.io/examples/channel_overrides.html

Thanks

hamishwillee, 2017-05-04:
Why do you want to use them? Highly NOT recommended.

The number are PWM values - the actual values sent out to the servos that drive the vehicle. So you'll have the normal RC values coming in from your receiver and going into the autopilot. Then the autopilot checks the channel and if you have an override sends the value you specify above instead. 

The channel numbers are (I believe) as you say "by default", but these can be changed by the autopilot. Normally they won't be. 

This just fakes an RC input. So if you send updates with the same PWM values that  you would require to live off as you have said above, that's what it will do. The only caveat is that the autopilot can only respond to messages at a certain rate. 

I'm not going to update the docs - we really don't want people using this feature.

toashwin, 2017-05-04:
As I said, the reason why I want to use channel_overrides is because: 

How do I control the Speed and time? For example I want lift for 1sec. and stop. I want to go very slow.
Can I lift off, for 0.1sec at the slowest speed possible?

Or after take off, I would like to adjust its height by a centimeter.. move in steps up and down..

hamishwillee, 2017-05-04:
I answered that. If you can do it manually with RC control then you can do it using channel overrides. I can't tell you exactly how to do it for your vehicle etc, because you're going to have to tune if for your throttle, vehicle, motors etc.

studroid, 2017-05-07:
This may help you: #712 
Channel override needs many specific tuning to properly control your drone, so changing attribute will be easier to apply for this.

EEMariam, 2018-02-24:
The channel override does not wok for me, the servos dont move.
The servos are powered properly and connected. The vehicle is disarmed.
I am using Raspberry Pi to send commands to MINI Pixhawk; I do execute the code but nothing happens
Any idea how to make it work?

angela804, 2022-07-31:
@EEMariam had my channel override work when the transmitter is turned on