dronekit/dronekit-python #714: Whats the right command to lift at the slowest speed. And also have precise control on altitude?

toashwin, 2017-05-03:
Hi

I am looking for drone_kit/Mavlink commands to use to achieve the following.

1) I want to lift off at the slowest speed possible. May be 1meter in 5sec. How can control that through script?
2) Once the drone reaches the right altitude, I would like to instruct done to go up or down only 1cm or 2cm and slowly.

Is  it possible to do it in dronekit? I was thinking RC_Overrides is the only way to do it?

squilter, 2017-05-03:
1. I think the best that you can do is just command a normal takeoff, but lower `WPNAV_SPEED_UP`. But this is probably a bad idea. You want to get off the ground quickly so that nothing gets snagged on the ground on takeoff. If you spend multiple seconds at only a couple cm off the ground, you will drift, and the landing gear could easily get stuck somewhere and flip over.

2. Yes, you can send position commands in guided mode. Although you can definitely send values that differ by only a few cm, I don't think the vehicle is accurate enough to actually do it.

toashwin, 2017-05-03:
What would be MAVLINK commands (or Dronekit ) to make the drone go up or down only couple of centimeters?

hamishwillee, 2017-05-03:
@toashwin - does your vehicle have a very accurate rangefinder? GPS will not give you centimeter precision? 
As to commands for moving, read the docs. They don't cover movement by centimetres, but you should be able to work it out: http://python.dronekit.io/