dronekit/dronekit-python #716: "Device /dev/ttyUSB1 is dead",

s4lvi, 2017-05-07:
I am trying to run this code:

```
from dronekit import connect

vehicle = connect('/dev/ttyUSB1', baud=57600)
print(vehicle.version)
print(vehicle.battery)
```

But I'm getting this output and not sure what the issue is. I tried disabling the prearm check, and it does thte samething except without the prearm messages and never returns a connection/vehicle. I was able to connect and download waypoints through mavproxy.py

Device /dev/ttyUSB1 is dead
>>> PreArm: Waiting for Nav Checks
>>> PreArm: Accels not calibrated
>>> PreArm: Waiting for Nav Checks
>>> PreArm: Accels not calibrated
>>> PreArm: Waiting for Nav Checks
>>> PreArm: Accels not calibrated
>>> PreArm: Waiting for Nav Checks
>>> PreArm: Accels not calibrated
>>> PreArm: Waiting for Nav Checks

peterbarker, 2017-05-07:
On Sun, 7 May 2017, Jordan Salvi wrote:

> But I'm getting this output and not sure what the issue is. I tried
> disabling the prearm check, and it does thte samething except without the
> prearm messages and never returns a connection/vehicle. I was able to
> connect and download waypoints through mavproxy.py

You need to calibrate your autopilot.

http://ardupilot.org/copter/docs/configuring-hardware.html

(assuming you have Copter on there)

s4lvi, 2017-05-08:
Shouldn't disabling the prearm check bypass this? When i do that i don't get any "not calibrated" messages but it never finishes connecting

s4lvi, 2017-05-15:
Calibration did end up allowing connection.

closed 2017-05-15 by s4lvi