dronekit/dronekit-python #719: Excessive Delay when polling vehicle.attitude, etc.

cheintz, 2017-05-30:
I am working on a multi-agent fixed wing UAV formation project. The agents share information over UDP broadcast via a raspberry pi v3B. In the main control thread, I poll various attributes (attitude, position, channels, etc) and am using roll angle as a test case for debugging.

I have found (by profiling) that the program spends the overwhelming majority of time waiting (to have a reasonable sampling time). This is expected. I have found that as the other threads consume resources (not many resources, but some), the reads, to, say, vehicle.attitude.roll produce values that are delayed by up to about 1.5 seconds. The resulting values are generally unique, but lag behind real-time by that amount.

I dug into the code, and it looks like a queue is being used in mavlink.py to get messages from pymavlink to dronekit. I suspect this queue is not being emptied fast enough. If this is, in fact, a likely cause, what is the cleanest way manually process all messages in this queue to get real-time data into the vehicle object?

ZixingLin, 2017-11-02:
Hi, did you manage to solve the delay issues? 

Also, can I ask how did you configure PX4 to send messages at a higher frequency? I am unable to find such a parameter.

cheintz, 2017-11-02:
My Delay issues were caused (I believe) by swamping my processor. This was
solved by using cPickle instead of jsonPickle to serialize some status
messages (there were and are a lot of big status messages)

Check out the SR0_positon and similar arduplane parameters (for the serial
port you're using) . I have them set to 20 (the ones I need). You may need
to increase the baud rate of the connection, using the appropriate
parameters on the pixhawk and arguments to your connect method.