dronekit/dronekit-python #720: Mission upload does work only the first time

GekoCH, 2017-06-05:
I use the example script from the dronekit page to upload a mission to the copter.
The first time it works great I can re-read the mission waypoints fine in the Missionplaner after they got uploaded through dronekit to the drone.
However when I re-upload the same mission again there is an additional waypoint at position 1 that before was not there. Therefore it is not possible to launch the mission

This is the mission file:
QGC WPL 110
0	1	0	16	0	0	0	0	47.388620	8.465068	446.828200	1
1	0	3	22	0.000000	0.000000	0.000000	0.000000	47.388817	8.464934	15.000000	1
2	0	3	82	0.000000	0.000000	0.000000	0.000000	47.388899	8.464759	15.000000	1
3	0	3	82	0.000000	0.000000	0.000000	0.000000	47.388711	8.464180	15.000000	1
4	0	3	82	0.000000	0.000000	0.000000	0.000000	47.388588	8.464708	15.000000	1
5	0	3	17	0.000000	0.000000	0.000000	0.000000	47.388670	8.464971	15.000000	1

This is how the mission looks when I read it from the Drone into MissionPlaner after the first upload:
<img>http://www.quadcam.ch/bilder/01.PNG</img>

And here after the second upload:
<img>http://www.quadcam.ch/bilder/02.PNG</img>

GekoCH, 2017-10-21:
hello...
anyone... this is really a problem and I don't have a workaround...

kaangoksal, 2018-09-08:
I also have the same problem currently, However the example file works fine somehow.

Hoggel0815, 2020-11-05:
Got no one a solution for that problem ?