dronekit/dronekit-python #724: Dronekit error on  PX4  -mode 67371008 not available on mavlink definition

hamishwillee, 2017-06-19:
PX4 dronekit example [here](https://dev.px4.io/en/dronekit/example.html) doesn't work. 

Specifically
1. Running on ubuntu in a VM
2. Dronekit set up as in that doc, from master. 
3. Using JMAVSIM SITL as per [Here](https://dev.px4.io/en/simulation/sitl.html). This is recent daily master, but problem has been reported for [months](http://discuss.px4.io/t/dronekitpx4-example-code-with-hils-is-not-working/2565) with SITL and HITL

The script connects and the vehicle takes off. But Dronekit gets stuck on first step takeoff. 

It reports
```
Executing mission
Exception in message handler for HEARTBEAT
mode 67371008 not available on mavlink definition
```
The PX4 build reports a [warning during startup](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L209) which I think is probably unrelated. "ERROR: wp index out of bounds"

peterbarker, 2017-06-19:
Agreed, the WP index is out-of-bounds is probably not related.

Make sure the autopilot is recognised as px4 by adding debug here: https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L1551

If it is crossing the PX4 codepath, then @squilter or @sanderux might be able to help by fixing `interpret_px4_mode` in pymavlink

sanderux, 2017-06-20:
This error seems to be related to a flightmode that is not recognized. This should have been fixed in the lates pymavlink. What versions dronekit and pymavlink are you running?

hamishwillee, 2017-06-20:
The latest you get by following the example instructions and rebuilding dronekit:
dronekit (2.9.1)
pymavlink (2.0.6)

sanderux, 2017-06-20:
Pymavlink should be 2.2.2 i believe, can you try upgrading?

hamishwillee, 2017-06-21:
Thanks @sanderux - pymavlink v2.2.3 works. Created https://github.com/dronekit/dronekit-python/pull/728 to up the number. @peterbarker can you merge if there is no reason for us "not" to do so.