dronekit/dronekit-python #729: PWM override

AhmedRejeb, 2017-06-21:
hi, my name is Ahmed, and I'm working on a Pixhawk with a raspberry board,
I developed a PID tuning loop on Simulink that have as output the PWM signal for  4 motors 
I want to use the DroneKit -python lib to send those messages (PWMs) to the Pixhawk to control the motor but I don't know how can you help me

peterbarker, 2017-06-21:
On Wed, 21 Jun 2017, AhmedRejeb wrote:

> hi, my name is Ahmed, and I'm working on a Pixhawk with a raspberry board,
> I developed a PID tuning loop on Simulink that have as output the PWM signal
> for 4 motors
> I want to use the DroneKit [...]

You will almost certainly find that the latencies mean flight will not 
be stable.  Consider ArduCopter's stabilization loops happen at 400Hz - 
that gives you 2.5millisecondss to get data to simulink, calculate the PWM 
outputs and return them to ArduCopter - that's really not going to happen 
from a latency perspective!

Additionally, I do not believe ArduPilot has the ability to source PWM 
information for the motors from any external source - however, you could 
notionally introduce a new GUIDED mode which takes information from a 
newly-crafted mavlink packet which sets the relevant outputs.

I can't talk to PX4 Pro's abilities to source motor PWM externally.

One option you may have is to talk directly to a simulator; there is a 
clear boundary between simulators and ArduPilot - sensor data from 
simulator to ArduPilot, pwm outputs from ArduPilot to simulator.  You may 
be able to use that to eliminate the autopilot altogether.  ArduPilot's
in-tree simulators are not the only ones with such clearly defined 
boundaries.

https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_Aircraft.cpp#L325
https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_Aircraft.h#L39

Using the ArduPilot simulator has the benefit of being able to 
lock-step-schedule to debug your code.

AhmedRejeb, 2017-06-22:
@peterbarker thanks for your quick response,
can you please explain to me more about this "however, you could
notionally introduce a new GUIDED mode which takes information from a
newly-crafted mavlink packet which sets the relevant outputs.""
and for the using simulator i can't see what is this can help me if I want truly to fly my copter not just simulate

peterbarker, 2017-06-24:
On Thu, 22 Jun 2017, AhmedRejeb wrote:

> @peterbarker thanks for your quick response,
> can you please explain to me more about this "however, you could
> notionally introduce a new GUIDED mode which takes information from a
> newly-crafted mavlink packet which sets the relevant outputs.""

We currently have two guided modes.  You could notionally add a third.

But seriously - consider the latencies involved!  Do a short feasibility 
though experiment before getting to invested in the concept.

> and for the using simulator i can't see what is this can help me if I want
> truly to fly my copter not just simulate

Simulating generally means less rebuilding :-)

AhmedRejeb, 2017-07-03:
thank Mr.@peterbarker for the help.
but now i have little different problems that I'm trying to solve but I can't:
I'm trying to make my drone fly and to move to certain distance by overriding RC commands, but the very first thing is I'm not able to arm my copter through [my script.](https://github.com/AhmedRejeb/DroneKitCode/blob/master/TakeOff2.py)
But its printing:
[image]
when i want to arm it the mavproxy like this .it work but when takeoff i have this error
[image]
Can you help me with this

peterbarker, 2017-07-04:
You need to be in guided for takeoff to work.  `mode guided` will do the trick - straigh tafter arming.

AhmedRejeb, 2017-07-04:
@peterbarker thanks for your quick response but it didn't work for me.
this what i did,
1-Stabilize mode  ->arm throttle -> takeoff 10
1- Stabilize mode  ->arm throttle ->mode guided --> takeoff 10
2-mode guided ->arm throttle -> takeoff 10 
and this the errors that i got:
[image]

i'm working in this code too but i can't make it work and arm my copter can you tell me what i'm doing wrong 
`#/usr/bin/env python
# -*- coding: utf-8 -*-
from dronekit import connect,VehicleMode,LocationGlobalRelative
from pymavlink import mavutil
import time
# Receiving from command line
import argparse
parser = argparse.ArgumentParser()
parser.add_argument('--connect',help="PORT_NO")
args = parser.parse_args()

# Connecting to the vehicle
connection_string = args.connect
print("Connecting to...% s" % connection_string)
vehicle = connect(connection_string, wait_ready=True)

#Function to arm and takeoff to a specified altitude
def arm_and_takeoff(aTargetAlt):
	print("Basic Prearm checks..dont touch!!")
	print("Waiting for vehicle to initialize")
	time.sleep(1)
	print("Arming Motors..")

	vehicle.mode = VehicleMode("STABILIZE")
	

	vehicle.armed = True
	vehicle.flush()
        print("Is Armed:% s"% vehicle.armed)
	
	while not vehicle.armed:
		print("Waiting for arming...Mode:% s"% vehicle.mode)
		print("Is Armed:% s"% vehicle.armed)
		vehicle.armed = True
		time.sleep(2)
		
	print("Taking Off..")
	vehicle.simple_takeOff(aTargetAlt)

	while True:
		print("Altitude: ",vehicle.location.global_relative_frame.alt)
		#Break and return from function just below target altitude.
		if vehicle.location.global_relative_frame.alt>=aTargetAlt: 
			print("Reached Target Altitude..")
			break
		time.sleep(2)
	
vehicle.close()
arm_and_takeoff(50)`
[image]

peterbarker, 2017-07-06:
You can't arm in guided mode.

suraj2596, 2019-12-02:
- Disable all pre-arm checks
- You either need a GPS lock or pixflow sensor
- set `mode LOITER`
- `arm safetyoff`
- `arm throttle`
- Then immediately give `takeoff 4`

Hope this helps!