dronekit/dronekit-python #731: api start Erreur

AhmedRejeb, 2017-07-01:
hello guys,
I'm new here and I hope that someone can help me to understand my I can't run a .py file after connecting throw mavlink to my pixhawk that rum APM firmware 
i tried to execute one of the dronekit example files like vehicle_state.py but when i connected using mavlink and after writing: api start vihicle_state.py, i got this error.

[image]

even after loading dronekit 
echo "module load droneapi.module.api" >> ~/.mavinit.scr
can some oe know why i'm having this prob ??

Karan02, 2017-11-25:
we are facing same problem

peterbarker, 2017-11-26:
dronekit-python hasn't been invoked this way for a very long time.

Which instructions were you following to get into this state?