dronekit/dronekit-python #734: Dronekit assumes first waypoint as home location

sanderux, 2017-07-11:
On APM the home location is stored as the first waypoint. This is not the case for PX4.
Throughout dronekit the assumption that the first waypoint (sequence 0) will be the home location is being implemented. It will even set this location on a clear().

This results in numerous issues when working with DroneKit on PX4.
One such issue is already reported here: https://github.com/dronekit/dronekit-python/issues/725

The code should be thoroughly sweeped and made autopilot aware. APM seems to be the exception here as the mavlink spec says nothing about special treatment of the first waypoint.

hamishwillee, 2017-07-11:
Agree.

GekoCH, 2017-10-19:
yep I got exactly the same problem see my issue here: https://github.com/dronekit/dronekit-python/issues/720

affourtit2, 2018-11-28:
Any updates on DroneKit and PX4 compatibility?

hamishwillee, 2018-11-29:
@affourtit2 There isn't any particular work going on to support PX4 compatibility other than making sure that it broadly works. More specifically, there is a problem if this does not work: http://dev.px4.io/en/robotics/dronekit.html

PX4 is now concentrating development effort on the Dronecode SDK. Most work on DroneKit is from interested community members.

Upshot, PRs are accepted, but don't expect that ti will become more compatible in any signficant way, any time soon.

affourtit2, 2018-11-29:
@hamishwillee I will investigate Dronecode SDK, then. Thank you for your response!

Guohao91, 2020-01-02:
@hamishwillee so is that possible to work fine by installing Ardupilot multiple flight stacks instead of PX4? I am using Pixhawk 4 controller.