dronekit/dronekit-python #735: Serial port is dead

BdT141, 2017-07-12:
Hello,

I'm trying to communicate between RPi Zero and Pixhawk 4 over the Pi's uart, but I get the error message "The device /dev/ttyS0 is dead" or /dev/serial0. I've tried looking in the souce code which function is giving that error, but can't find it. The serial port isn't the problem because I've tested it with other devices and it works. I was wondering if the problem is on the Pi's side or the Pixhawk's side.

Thank,
Brandon

BdT141, 2017-07-17:
Problem was I was trying to use Python3.

closed 2017-07-17 by BdT141

peterbarker, 2017-07-17:
On Mon, 17 Jul 2017, Brandon du Toit wrote:

> Problem was I was trying to use Python3.

Thanks for the update.  We'll now know more next time we see the same 
symptoms!

GTSyu, 2021-04-06:
> Problem was I was trying to use Python3.

Exceuse me. So could you please share how did you solve the problem?? did you reinstall mavlink and the dronekit in python2 version?