dronekit/dronekit-python #738: Remove hard dependency versions

sanderux, 2017-07-23:
This has been an issue for quite some time:
https://github.com/dronekit/dronekit-python/blob/master/setup.py#L14

Is there any reason not to set it as a minimum? 
For proper PX4 support the minimum should be 2.2.3

@hamishwillee @peterbarker

hamishwillee, 2017-07-24:
Nope, please update. We already did [requirments.txt](https://github.com/dronekit/dronekit-python/blob/master/requirements.txt) - no idea why setup.py would be different

peterbarker, 2017-07-24:
On Sun, 23 Jul 2017, Sander Smeets wrote:

> This has been an issue for quite some time: 
> https://github.com/dronekit/dronekit-python/blob/master/setup.py#L14
> 
> Is there any reason not to set it as a minimum?
> For proper PX4 support the minimum should be 2.2.3

This is an inconsistency between requirements.txt and setup.py; pushed in 
2.2.3 to make them consistent.