dronekit/dronekit-python #742: SAFETY CRITICAL: vehicle_state.py arms and runs motor at full reverse thrust if run on plane

cheintz, 2017-07-26:


peterbarker, 2017-07-27:
Thanks for the report.  Arming has been removed from this example - it only has to catch one person out to be worth removing!

Still a really bad idea to run this on a a real vehicle because it plays with throttle parameters(!), which could be a surprise later.  Not sure *who* thought that was a good idea!

closed 2017-07-27 by peterbarker

hamishwillee, 2017-07-27:
@peterbarker I'd rather disable the parameter setting code (ie comment it out with a note that parameters "might" be set like this). 
In any case, if you remove arming, then the "How to" instructions should  also be updated in the docs, since they assume you can just kick the code off. 

> Not sure who thought that was a good idea!

Someone incredibly attractive and clever. I'm certain of it :-)

cheintz, 2017-07-27:
Ha, ha. I had run the program on a PX4 on the bench before with no ill effects. I suspect the lack of RC,GPS,etc. caused it to not progress.

In the field, I ran it on a real plane to verify the SYSID_THISMAV, making the (admittedly idiotic) assumption that the script was read only based on the file name vehicle_state.py,  but was surprised when the motor went crazy on the bench.

I discovered the parameter changes later when, on the launcher, switching to FBWA for preflight checks caused the motor to run. I hastily disarmed, which did not stop the 

Fortunately, I had a backup of the parameters and nothing was hurt except my pride. I expect a long recovery.