dronekit/dronekit-python #747: ADSB_VEHICLE message is not available

drumadrian, 2017-08-24:
Hello DroneKit Developers, 

I have installed a Uavionix Ping Rx ADS-B receiver on my Pixhawk 2.1 and I can see the ADSB_VEHICLE message in QGroundControl but not using my DroneKit python code. 

The Ping Receiver:
[pingRX ADS-B Receiver](https://www.uavionix.com/products/pingrx/)

The MAVLINK message: 
[ADSB_VEHICLE](http://mavlink.org/messages/common#ADSB_VEHICLE)

I used the decorator to create a callback but it did not execute.  

I tested the decorator feature to run for all messages using the `* ` parameter and I did not see any ADSB_VEHICLE messages.  

ADS-B Code snippet: 

`#Create a message listener using the decorator.
@vehicle.on_message('ADSB_VEHICLE')
def listener(self, name, message):
    print '\n\n ADSB_VEHICLE message: %s \n\n' % message
`

Non ADS-B Code snippet: 

`
#Create a message listener using the decorator.
@vehicle.on_message('*')
def listener(self, name, message):
    print '\n\n\nReceived Non-ADS-B message named: %s\n\n\n' % name
`

_I hope I am reporting this issue correctly.  I didn't see any issues regarding ADS-B in the repo.  It is my first for this project/repo and I'm happy to provide more information or develop a fix if I get a little help and guidance._ 

Thanks, 

_Adrian_

peterbarker, 2017-08-24:
On Wed, 23 Aug 2017, Adrian wrote:

> I tested the decorator feature to run for all messages using the * parameter
> and I did not see any ADSB_VEHICLE messages.

Are your pymavlink message bindings up-to-date?

drumadrian, 2017-08-24:
Hi, I'm still a little new to DroneKit. How can I ensure that I have up to date pymavlink bindings?

Thanks for the quick reply,

Adrian

peterbarker, 2017-08-24:
On Thu, 24 Aug 2017, Adrian wrote:

> Hi, I'm still a little new to DroneKit. How can I ensure that I have up to
> date pymavlink bindings?

Usually I just install them from a clone github repository using

MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions python 
setup.py build install --user --force

drumadrian, 2017-08-26:
Hi Peter, I'm not exactly sure what to do, but do you mean this?

1) Clone the pymavlink repo from here: 
https://github.com/ArduPilot/pymavlink

2) Setup the following environment variable: 
     MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions

3) Run this setup.py file with these arguments(build install --user --force):
https://github.com/ArduPilot/pymavlink/blob/d73cee90fa7a598230b19478bfcc58e51f84610b/setup.py

Did I understand this correctly?  

Sorry about the blog-like questions.  I wasn't sure where else to report this and get help.  I really want to contribute to ADS-B usage on Pixhawk.  

Thank you, 

Adrian

peterbarker, 2017-08-29:
@drumadrian Do you know how to run MAVProxy?  This will be an easy way to check your Python bindings, and I should have suggested it first.

`mavproxy.py` followed by `status ADSB_VEHICLE`

drumadrian, 2017-08-29:
Oh cool, let me check that now.  

I usually run MAVProxy by starting my python virtual environment, then running the python program located here: 

[https://github.com/drumadrian/adsb-drone-code/blob/master/testMAVLINK_NO_CREDENTIALS.py](https://github.com/drumadrian/adsb-drone-code/blob/master/testMAVLINK_NO_CREDENTIALS.py)

the command line argument looks like this: 

`python testMAVLINK.py --connect '/dev/ttyMFD1,921600'
`

I will try the command you just mentioned with mavproxy running right now and reply with the results.

drumadrian, 2017-08-30:
I just tried to run MAVProxy and it did not work. 

When i tried to run it without any arguments, it's prompted with this and tried to connect to the simulator: 

> Auto-detected serial ports are:
Connect 0.0.0.0:14550 source_system=255
Failed to load module: No module named terrain. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Failed to load module: No module named adsb. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from 0.0.0.0:14550
 MAV> 

I have my PingRx connected on the Telem2 port so I used this command to start MAVProxy:

> mavproxy.py --master=/dev/ttyMFD1

I'm confident that I am using the correct device path "/dev/ttyMFD1" since I have successfully run 
vehicle_state.py

A sample of that output is here: 
> (drone-virtualenv) root@jubilinux:# python mavproxy.py --master=/dev/ttyMFD1 baudrate 921600
python: can't open file 'mavproxy.py': [Errno 2] No such file or directory
(drone-virtualenv) root@jubilinux:~# python vehicle_state.py --connect /dev/ttyMFD1,921600
Connecting to vehicle on: /dev/ttyMFD1,921600
>>> APM:Copter V3.5.0 (633501f9)
>>> PX4: 33825946 NuttX: 1a99ba58
>>> Frame: HEXA
>>> PX4v3 00420022 30365119 30393333
Get all vehicle attribute values:
 Autopilot Firmware version: APM:UnknownVehicleType13-3.5.0
   Major version number: 3
   Minor version number: 5
   Patch version number: 0
   Release type: rc
   Release version: 0
   Stable release?: True

I'm not sure what to do right now.  I'm willing to start over from scratch, but I don't know what isn't working and how to fix it.  

Adrian

drumadrian, 2017-08-30:
I just reinstalled dronekit and I received this output when i tried to run MavProxy again:

> (drone-virtualenv) edison@jubilinux:/AWScode$ mavproxy.py --master=/dev/ttyMFD1
Connect /dev/ttyMFD1 source_system=255
Failed to load module: No module named terrain. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Failed to load module: No module named adsb. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Log Directory: 
Telemetry log: mav.tlog
ERROR: Not enough free disk space for logfile
Waiting for heartbeat from /dev/ttyMFD1
S+7 Unloading module link
Unloading module log
Unloading module signing
Unloading module wp
Unloading module rally
Unloading module fence
Unloading module param
Unloading module relay
Unloading module tuneopt
Unloading module arm
Unloading module mode
Unloading module calibration
Unloading module rc
Unloading module auxopt
Unloading module misc
Unloading module cmdlong
Unloading module battery
Unloading module output
(drone-virtualenv) edison@jubilinux:~/AWScode$

drumadrian, 2017-08-30:
I keep looking at this error and I realized that I need to run the script from inside of the 'pymavlink' folder: 

> IOError: [Errno 2] No such file or directory: 'message_definitions/v1.0/ardupilotmega.xml'

I did that and the script started running again.  

Adrian

peterbarker, 2017-08-30:
On Tue, 29 Aug 2017, Adrian wrote:

> I have my PingRx connected on the Telem2 port so I used this command to
> start MAVProxy:
>
>       mavproxy.py --master=/dev/ttyMFD1

Add the baudrate:

mavproxy.py --master /dev/ttyMFD1 --baudrate 921600

drumadrian, 2017-08-30:
I just tried using the --baudrate 921600 parameter and it failed.  This is the output: 

> (drone-virtualenv) edison@jubilinux:/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink$ mavproxy.py --master /dev/ttyMFD1 --baudrate 921600
Connect /dev/ttyMFD1 source_system=255
Failed to load module: No module named terrain. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Failed to load module: No module named adsb. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Log Directory: 
Telemetry log: mav.tlog
ERROR: Not enough free disk space for logfile
Unloading module linkWaiting for heartbeat from /dev/ttyMFD1
Unloading module log
Unloading module signing
Unloading module wp
Unloading module rally
Unloading module fence
Unloading module param
Unloading module relay
Unloading module tuneopt
Unloading module arm
Unloading module mode
Unloading module calibration
Unloading module rc
Unloading module auxopt
Unloading module misc
Unloading module cmdlong
Unloading module battery
Unloading module output
(drone-virtualenv) edison@jubilinux:~/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink$ 

I also just re-ran the build command you originally gave me and I got this error.  Now I'm not sure what to do:

>(drone-virtualenv) edison@jubilinux:/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink$ python setup.py build install --user --force
running build
running build_py
Using message definitions from /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/slugs.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/paparazzi.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/minimal.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/ualberta.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/ASLUAV.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/ardupilotmega.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/autoquad.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/test.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/python_array_test.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/uAvionix.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/matrixpilot.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/standard.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/common.xml for protocol 1.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/slugs.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/paparazzi.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/minimal.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/ualberta.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/ASLUAV.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/ardupilotmega.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/autoquad.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/test.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/python_array_test.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/uAvionix.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/matrixpilot.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/standard.xml for protocol 2.0
Building /home/edison/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink/message_definitions/v1.0/common.xml for protocol 2.0
creating build
creating build/lib.linux-i686-2.7
creating build/lib.linux-i686-2.7/pymavlink
copying ./mavutil.py -> build/lib.linux-i686-2.7/pymavlink
copying ./quaternion.py -> build/lib.linux-i686-2.7/pymavlink
copying ./mavextra.py -> build/lib.linux-i686-2.7/pymavlink
[... cut ...]
i586-linux-gnu-gcc: error: mavnative/mavnative.c: No such file or directory
i586-linux-gnu-gcc: fatal error: no input files
compilation terminated.
error: command 'i586-linux-gnu-gcc' failed with exit status 4
(drone-virtualenv) edison@jubilinux:/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink$ i586-linux-gnu-gcc
i586-linux-gnu-gcc: fatal error: no input files
compilation terminated.
(drone-virtualenv) edison@jubilinux:/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink$

drumadrian, 2017-08-30:
I am using jubilinux on an Intel Edison.  I hope this low free disk space isn't causing an issue.  I don't know what else is safe to delete to free up space.  

here is the disk space situation: 

> (drone-virtualenv) edison@jubilinux:/AWScode/drone-virtualenv/lib/python2.7/site-packages/pymavlink$ df -h
Filesystem       Size  Used Avail Use% Mounted on
/dev/root        1.4G  455M  861M  35% /
devtmpfs         480M     0  480M   0% /dev
tmpfs            481M     0  481M   0% /dev/shm
tmpfs            481M   13M  468M   3% /run
tmpfs            5.0M     0  5.0M   0% /run/lock
tmpfs            481M     0  481M   0% /sys/fs/cgroup
tmpfs            481M     0  481M   0% /tmp
/dev/mmcblk0p10  1.3G  1.1G  178M  87% /home
/dev/mmcblk0p7    32M  5.1M   27M  16% /boot
tmpfs             97M     0   97M   0% /run/user/1002