dronekit/dronekit-python #753: New Feature Proposal: Optical Flow and Rangefinder for indoor missions

rogerwwww, 2017-09-18:
Hi, I have succeeded in guiding my copter with **Optical Flow** and **Rangefinder** indoors, and I had several successful flights ([video here](https://youtu.be/HeJs-0_Bv8s)). My code is available [here](https://github.com/Drones4STEM/DroneController/tree/master/GUIDED).

I've modified the existing pre-flight checks to suite the OF+RF indoor environment. I've created [a subclass from ``Vehicle``](https://github.com/Drones4STEM/DroneController/blob/master/GUIDED/FlightController.py#L53), added a message listener for ``EKF_PRED_POS_HORIZ_REL`` and overloaded the ``is_armable`` attribute in class ``Vehicle``. 

Are you interested in this new feature? Do you have any suggestions?

Thank you!

rogerwwww, 2017-09-19:
Sorry I did not make myself clear: I am interested in integrating my code into dronekit to give it the new feature. Should I make a PR directly or do something else?

hamishwillee, 2017-09-19:
@rogerwwww This is pretty cool, and I know that there has been a lot of interest in this in the community. You should certainly let people know about this on our gitter channel and discuss forum!

I'm not particularly involved in DK-Python any more, but some thoughts:
1. Deriving from base vehicle class is the right way to add this sort of feature. 
2. This does not have to be part of DK-Python proper - the class can live in its own file and people can use it directly - as per http://python.dronekit.io/examples/create_attribute.html 
2. The name of the classes aren't great though - "Custom_DroneKit_Vehicle" would be better as perhaps Indoor_DroneKit_Vehicle. 

I don't know how to go about getting something like this integrated into the "core" or whether this is exactly the right way to do that. One option is to update the documentation to have a section on indoor use, and point to your class as a possible implementation.

@peterbarker @mrpollo - you might want to comment.

mrpollo, 2017-09-19:
I see a lot of base customizations on your source directory, I would welcome any changes that would extend the base `Vehicle` class in such way that won't break current functionality.

@rogerwwww why don't you send a PR with the changes you are proposing, we can talk more there.

rogerwwww, 2017-09-20:
@hamishwillee @mrpollo Thanks for your suggestions! I will modify my code into the style of current DroneKit-Python examples and send a PR with the new changes. And I will let the community know on our gitter and forum😃

hamishwillee, 2017-09-20:
@rogerwwww Great. 
Actually, what @mrpollo  is suggesting is that you integrate your changes into the core DK-Python in the `Vehicle` base class - provided you can do so in a way that is compatible with/ won't break existing functionality.  Once you have a PR we can discuss the changes to verify they don't break anything. 

The benefit is that any compatible changes then are more easily available to all sub classes, without having to discover your enhancements.

rogerwwww, 2017-09-20:
@hamishwillee Got it. I will make a PR in a few days. Thanks!

rogerwwww, 2017-09-21:
A PR was sent.

closed 2017-09-21 by rogerwwww