dronekit/dronekit-python #755: Connection issue

asiyo, 2017-09-26:
Hello
I am new to dronekit and I have some trebles when I try to connect my drone via dronekit-python.
I have no trebles when I'm using QGC (wifi and usb)
I'm using :

- pixracer with PX4 
- python 2.7
- dronekit-python (don't know witch version but I downloaded it few days ago via pip)
- windows 7 and 10 (2 different PC)

when I try to run the most basic code:
```
from dronekit import connect
connection_string = '127.0.0.1:14550'
print "Connecting"
vehicle = connect(connection_string, wait_ready=True)
```
I get the following output:

> Connecting
> >>> Link timeout, no heartbeat in last 5 seconds
> >>> No heartbeat in 30 seconds, aborting.
> Traceback (most recent call last):
>   File "hello1.py", line 6, in <module>
>     vehicle = connect(connection_string, wait_ready=True)
>   File "build\bdist.win-amd64\egg\dronekit\__init__.py", line 2880, in connect
>   File "build\bdist.win-amd64\egg\dronekit\__init__.py", line 2147, in initialize
> dronekit.APIException: Timeout in initializing connection.

Can someone tell me what am I doing wrong and walk me throw.
 
Thank you in advance.

hamishwillee, 2017-09-26:
Suggest you try the gitter channel. I'd use DK-python from github rather than pip.

The message means what it says - you have no connection. The connection string is for a UDP connection to local host on port 14550 - that works for SITL but not for real hardware. You will need to specify the actual connection string for the method you're using to connect.

Last of all, I am not sure how well DK-Python is currently handling PX4. I suggest you look at [DroneCore](http://dronecore.io/). It is C++ and still evolving, but it is designed for PX4 and should have Python wrappers in the coming months.

asiyo, 2017-09-27:
@hamishwillee **Thank you for your quick answer!**

1) Can you tell me how can I find the right connection string? (for now I'm connecting to the pixracer directly from my laptop via wifi or USB (e.g COM5 )).
2)Is DK-Python more suited to arducopter? 
3) I tried to install arducopter (via QGC and MP) and I failed to do it. Is it possible to install arducopter on pixracer?

Thank you in advance.

hamishwillee, 2017-09-27:
Sorry, no. I am off on holiday. Back in a couple of week. Raise issue on gitter channel.

guglie, 2017-10-04:
> Can you tell me how can I find the right connection string? (for now I'm connecting to the pixracer directly from my laptop via wifi or USB (e.g COM5 )).

I think the connection string is as simple as "COM5" in your example. Don't forget to set the baud rate.

To install DK-python from repository:
```
git clone https://github.com/<your_fork_of_dronekit>/dronekit-python.git
cd dronekit-python  
python setup.py build
python setup.py install
```

yoshiioko, 2018-01-08:
Hi asiyo,

I was having the same issue as you when trying to run a program to connect to my device (Intel Aero drone). FYI, I ran the program from within the Aero drone itself. Had to make these changes:

1. Updated the main.conf file for the mavlink-router with an additional UDP endpoint by adding these lines:
`[UdpEndpoint dronekit]`
`Mode = Normal`
`Address = 127.0.0.1`
`Port = 11000`

2. Restarted mavlink-router service:
`systemctl restart mavlink-router`

3. Changed connect() to just do this:
`vehicle = connect('127.0.0.1:11000')`

I am not an expert! But I can now get some vehicle state data from the system.