dronekit/dronekit-python #757: play_tune example does not work

GekoCH, 2017-10-12:
Hy

the play_tune example does not work with the dronekit version 2.9.1 is this there not yert implemented?
`AttributeError: 'Vehicle' object has no attribute 'play_tune'`

Andy

peterbarker, 2017-10-12:
On Thu, 12 Oct 2017, GekoCH wrote:

> the play_tune example does not work with the dronekit version 2.9.1 is this
> there not yert implemented?
> AttributeError: 'Vehicle' object has no attribute 'play_tune'

Some other dronekit-python is in your path.  If the example is present in 
a checkout then the vehicle will have the code for it...

GekoCH, 2017-10-12:
hmm I did install the dronekit over `pip install` do I need to clone the GIT and use it that way?

peterbarker, 2017-10-12:
On Thu, 12 Oct 2017, GekoCH wrote:

> hmm I did install the dronekit over pip install do I need to clone the GIT
> and use it that way?

Yes.  And pip uninstall the pip-installed version to avoid confusion.

I assume you grabbed the demo from the internet, then?

GekoCH, 2017-10-12:
I did that and now did the same thing with the MAVproxy everything is now loaded by GIT and installed but still now I get this error:
`AttributeError: 'MAVLink' object has no attribute 'play_tune_encode'`

peterbarker, 2017-10-12:
On Thu, 12 Oct 2017, GekoCH wrote:

> I did that and now did the same thing with the MAVproxy everything is now
> loaded by GIT and installed but still now I get this error:
> AttributeError: 'MAVLink' object has no attribute 'play_tune_encode'

Sorry, now you need to update your pymavlink :-)

You can do this in much the same way as you did dronekit-python, with the 
exception that you need to tell setup.py where to find the message 
definitions.  I usually do this with:

MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions python 
setup.py build install --user --force

GekoCH, 2017-10-13:
sorry to disturb you once more...

ok I did now clone `https://github.com/ArduPilot/pymavlink` and `https://github.com/ArduPilot/mavlink`

the I did run the `python setup.py build` and `sudo python setup.py install` command in den pymavlink folder 

then I used the command `MDEF=$HOME/xxx/mavlink/message_definitions`
and `python setup.py build install --user --force` in the pymavlink folder

but I still get the error message `AttributeError: 'MAVLink' object has no attribute 'play_tune_encode'`

what am I doing wrong?

peterbarker, 2017-10-13:
On Thu, 12 Oct 2017, GekoCH wrote:

> but I still get the error message AttributeError: 'MAVLink' object has no
> attribute 'play_tune_encode'
> 
> what am I doing wrong?

Did you remove the pip-installed version of pymavlink?

Yeep removing things with pip 'til it starts telling there's none left, 
then install from the reo....

GekoCH, 2017-10-13:
i did remove but now when I look for the new installed by using `pip list` there is no pymavlink
even the code posted above finished with 
`Finished processing dependencies for pymavlink==2.2.5`

APMLover, 2017-10-14:
Im suffering this issues too. it worked well with JTK1 board but with new raspberry it doesnt work

sgh055, 2017-10-19:
play_tune function doesn't work in raspberry Pi3
[image]
how fix it?

sgh055, 2017-10-19:
I try mavlink message...
it doesn't work
[image]
no attribute play_tune / play_tune_encode

tizianofiorenzani, 2018-04-23:
So, it looks like you need to set mavlink2 typing  MAVLINK20=1. I set it up in rc.local

YusufYigitbasi, 2018-06-30:
On linux ubuntu
i tested dronekit-sitl copter simulation simple_go_to.py when i entered "mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551 --map --console"
on terminal it gave this crush
Terminal map and console connection error :

 Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/yusuf/.local/bin/mavproxy.py", line 783, in main_loop
    master.wait_heartbeat()
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 383, in wait_heartbeat
    return self.recv_match(type='HEARTBEAT', blocking=blocking)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 344, in recv_match
    m = self.recv_msg()
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/mavutil.py", line 318, in recv_msg
    msg = self.mav.parse_char(s)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9257, in parse_char
    self.__callbacks(m)
  File "/usr/local/lib/python2.7/dist-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9234, in __callbacks
    self.callback(msg, *self.callback_args, **self.callback_kwargs)
  File "/home/yusuf/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_link.py", line 460, in master_callback
    mavutil.mavlink.MAV_TYPE_DODECAROTOR]:
AttributeError: 'module' object has no attribute 'MAV_TYPE_DODECAROTOR'
then when i wanted to run simple_go_to.py it gave timeout error:
Link timeout, no heartbeat in last 5 seconds
>>> No heartbeat in 30 seconds, aborting.
Traceback (most recent call last):
  File "simple_goto.py", line 38, in <module>
    vehicle = connect(connection_string, wait_ready=True)
  File "/home/yusuf/.local/lib/python2.7/site-packages/dronekit/__init__.py", line 2845, in connect
    vehicle.initialize(rate=rate, heartbeat_timeout=heartbeat_timeout)
  File "/home/yusuf/.local/lib/python2.7/site-packages/dronekit/__init__.py", line 2117, in initialize
    raise APIException('Timeout in initializing connection.')
dronekit.APIException: Timeout in initializing connection.

pip version is 8.1.1
python : 2.7
mavlink 2.0.6
what should i do i saw? someone said that upgrade mavlink to 2.2.8 but i couldnt do that 
thanks for help :)

peterbarker, 2018-07-04:
On Sat, 30 Jun 2018, YusufYigitbasi wrote:

> what should i do i saw? someone said that upgrade mavlink to 2.2.8 but i couldnt do that

Upgrading that is the solution.

Why can't you upgrade?

tizianofiorenzani, 2018-10-12:
The current implementation of vehicle.play_tunes is not compatible with the last Mavlink protocol. An extra argument is expected. Here it is my simple patch in the code:
                    if vehicle.version.major == 3 and vehicle.version.minor <= 5 and vehicle.version.patch <=5:
                        # print ("Older than 3.5.5 release")
                        vehicle.play_tune(tune_to_play)
                    else:
                        # print ("New release")
                        msg = vehicle.message_factory.play_tune_encode(0, 0, tune_to_play, tune_to_play)
                        vehicle.send_mavlink(msg)
                    # print ("-- Tune Played --")

peterbarker, 2018-10-12:
On Fri, 12 Oct 2018, Tiziano Fiorenzani wrote:

> The current implementation of vehicle.play_tunes is not compatible with the last Mavlink protocol. An extra argument is expected. Here it is my simple

What error do you get with the old code?

peterbarker, 2018-10-15:
Please note your autopilot has to speaking mavlink2 for `play_tune_encode` to appear.

That's `param set SERIAL0_PROTOCOL 2` in MAVProxy.

peterbarker, 2018-10-15:
On Fri, 12 Oct 2018, Tiziano Fiorenzani wrote:

> Something like "s must be a string"

OK, so I reproduced that easily enough, thanks.

I've a concern about the patch you're suggesting here, however.  It will 
certainly fix the test - but it's a workaround.

I think this issue might arise any time a message has extensions; we need 
to come to grips with the wider problem, I think.

peterbarker, 2018-10-15:
https://github.com/ArduPilot/pymavlink/blob/master/generator/mavgen_python.py#L315 <- that's the problem line.

peterbarker, 2018-10-15:
https://github.com/ArduPilot/pymavlink/pull/227 <- I believe that might be the correct fix.

tizianofiorenzani, 2018-10-15:
> ArduPilot/pymavlink#227 <https://github.com/ArduPilot/pymavlink/pull/227>
> <- I believe that might be the correct fix.

Thanks Peter, the suggestion for patching was very quick and dirty, more
for giving an idea of the problem rather that been a real solution. I will
test it right away

tizianofiorenzani, 2018-10-15:
Just uninstalled pymavlink and reinstalled through dronekit (went from 2.2.15 to 2.2.16). I still get  "argument for 's' must be a string" when I call play_tunes. MAVLINK20 is set to 1

peterbarker, 2018-10-15:
On Mon, 15 Oct 2018, Tiziano Fiorenzani wrote:

> Just uninstalled pymavlink and reinstalled through dronekit (went from 2.2.15 to 2.2.16). I still get "argument for 's' must be a string" when I call
> play_tunes. MAVLINK20 is set to 1

The patch is still in the PR I referenced - not even merged yet, let alone 
released into the wild :-)

peterbarker, 2018-10-16:
On Tue, 16 Oct 2018, Peter Barker wrote:

> The patch is still in the PR I referenced - not even merged yet, let alone 
> released into the wild :-)

The PR has now been merged.

Next pymavlink release will have the fix.

In the meantime you can run from the master branch if you need the fix :-)

sergiimokin, 2024-08-06:
I am using pymavlink version 2.4.41 and have installed mavlink from the main branch of https://github.com/mavlink/mavlink.git.

However, I am still encountering the following error:

```
Traceback (most recent call last):
  File "/home/opi/ws/mavlink/play_tune.py", line 41, in <module>
    vehicle.play_tune(args.tune)
  File "/home/opi/.venv/lib/python3.11/site-packages/dronekit/__init__.py", line 2362, in play_tune
    msg = self.message_factory.play_tune_encode(0, 0, tune)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'MAVLink' object has no attribute 'play_tune_encode'. Did you mean: 'altitude_encode'?
```

I have carefully reviewed the related topic from 2018, and I believed this issue had been resolved. However, the problem still persists for me.

Could you please provide any guidance or updates on how to resolve this issue?

Thank you.