dronekit/dronekit-python #759: "Device /dev/ttyACM0 is dead" with python3

technorior, 2017-10-15:
APM: Copter v3.5.3
when I use:
`vehicle = connect("/dev/ttyACM0", boud=ANY)`
in python 2 work great, but in python 3 I have message:

> Device /dev/ttyACM0 is dead

and then

> >>>> No IO Thread Heartbeat

Evidlo, 2017-12-14:
I have this problem too, without the `No IO Thread Heartbeat` part.

pietrodn, 2018-01-11:
I have the same problem of @technorior with PX4 firmware.

peterbarker, 2018-01-11:
@Evidlo What interactive session?  dronekit-python doesn't do those naturally.

@technorior Try using `/dev/serial/by-id/<YOURDEVICENAME>` instead of `/dev/ttyACM0` - you're more likely to be able to reconnect to that if the device reboots.

@technorior The "No IO Thread Heartbeat" is probably because you don't have an SD card inserted.

pietrodn, 2018-01-12:
I fixed this problem by [patching pymavlink](https://github.com/ArduPilot/pymavlink/pull/137).

Evidlo, 2018-01-15:
I no longer have a Pixhawk, so I can't confirm.  It doesn't happen when connecting to the simulator.

@peterbarker All I meant was that `connect()` blocks forever.

pietrodn, 2018-01-22:
Should be fixed in master.

otsuka-kohei, 2018-12-20:
I found same problem on ArduPilot Copter 3.6.3, Python 3.7.1, Windows 10, and dronekit (latest github repo master).
```
Device com3 is dead
Exception in MAVLink input loop
Traceback (most recent call last):
  File "C:\Users\otsuka\AppData\Local\Programs\Python\Python37-32\lib\site-packages\dronekit\mavlink.py", line 210, in mavlink_thread_in
    fn(self)
  File "C:\Users\otsuka\AppData\Local\Programs\Python\Python37-32\lib\site-packages\dronekit\__init__.py", line 1370, in listener
    self._heartbeat_error)
dronekit.APIException: No heartbeat in 30 seconds, aborting.
```