dronekit/dronekit-python #761: PX4 ArduPlane  / Raspberry Pi subsequent connect issue (USB)

cheintz, 2017-10-24:
I am connecting to a PX4 running ArduPlane, I connect via USB from a Raspberry Pi3. My script sends RC overrides and reads the general vehicle state variables (speed, position, mode, RC, etc). I connect with the line: 

stdbuf -oL python flightProgram.py --connect /dev/ttyACM0 | tee -i $LOGPATH/$outFile

flightProgram.py uses the following line: 
vehicle = connect(connection_string, wait_ready=True, rate=20, baud=1500000,vehicle_class=ServoVehicle), 

where ServoVehicle is a superclass of vehicle that keeps track of RC output values as well.

Code here: https://github.com/cheintz/DroneKit_FW_Formation2017/tree/development/flightControl

If I stop and then restart my script, the connection fails. I get zero to maybe 10 re-connects before having to powercycle the Pi and PX4. This is inconvenient as it requires landing.

This problem does NOT occur when I connect my script to a simvehicle.py simulated vehicle in Ubuntu virtual machine. 

Can you all think of anything that would be causing this? Do you think using one of the hardware serial ports on the PX4 and Pi would help?

Thanks,
Chris

VarunNelakanti, 2018-04-19:
Please close the connection by using "vehicle.close()" command when you are stopping the script. The reason it could not connect was because the connection was still open according to Pixhawk. I also faced the same issue and by giving the close() command, never faced it again

cheintz, 2018-04-19:
Thank you for the suggestion! Unfortunately, I already call "vehicle.close()", and still suffer the problem. For some reason, it is difficult to reproduce on the bench, but always seems to happen when I need to reconnect to the vehicle while it is in flight.

VarunNelakanti, 2018-04-20:
Have you tried to use the Telemetry2 port instead of USB post in the Pixhawk? USB, while it can be used, is not recommended at all to connect while using any type of companion computer. They have put in a USB port simply so that people can easily program and connect to the Pixhawk.

cheintz, 2018-04-20:
I am in the process of testing that, though it's on the back burner for
now. Again, it's hard to replicate the test on the bench.

As an aside, do you have any insight into how to get data through telemetry
2 faster than 10hz?