dronekit/dronekit-python #769: Errors when trying to override the channels

liurenju, 2017-11-14:
hi,

I'm trying to use python 3.5 to run the example channel overriding script:
https://github.com/dronekit/dronekit-python/blob/master/examples/channel_overrides/channel_overrides.py

But it gave me an error of in the dronekit __init__ func:
`
Traceback (most recent call last):
  File "channel_overrides.py", line 66, in <module>
    vehicle.channels.overrides['2'] = 200
  File "/usr/local/lib/python3.5/dist-packages/dronekit/__init__.py", line 741, in __setitem__
    self._send()
  File "/usr/local/lib/python3.5/dist-packages/dronekit/__init__.py", line 753, in _send
    for k, v in self.iteritems():
AttributeError: 'ChannelsOverride' object has no attribute 'iteritems'
`
Does anyone know the way to fix the issue? Thanks

liurenju, 2017-11-14:
I fixed the above issue by change iteritems() to items() for python3.5, but it doesn't return me any data.

Channel values from RC Tx: {'1': None, '2': None, '4': None, '7': None, '5': None, '6': None, '8': None, '3': None}
Read channels individually:
 Ch1: None
 Ch2: None
 Ch3: None
 Ch4: None
 Ch5: None
 Ch6: None
 Ch7: None
 Ch8: None
Number of channels: 8

Channel overrides: {}
Set Ch2 override to 200 (indexing syntax)
 Channel overrides: {'2': 200}
 Ch2 override: 200
Set Ch3 override to 300 (dictionary syntax)
 Channel overrides: {'3': 300}
Set Ch1-Ch8 overrides to 110-810 respectively
 Channel overrides: {'1': 110, '2': 210, '4': 4100, '3': 310, '7': 710, '6': 610, '8': 810, '5': 510}

Cancel Ch2 override (indexing syntax)
 Channel overrides: {'1': 110, '4': 4100, '3': 310, '7': 710, '6': 610, '8': 810, '5': 510}
Clear Ch3 override (del syntax)
 Channel overrides: {'1': 110, '4': 4100, '7': 710, '6': 610, '8': 810, '5': 510}
Clear Ch5, Ch6 override and set channel 3 to 500 (dictionary syntax)
 Channel overrides: {'3': 500}
Clear all overrides
 Channel overrides: {}

Close vehicle object
Completed

liurenju, 2017-11-14:
Need to set the mode to be manual.

vehicle.mode = VehicleMode("MANUAL")

closed 2017-11-14 by liurenju

kanhao100, 2021-03-14:
> 需要将模式设置为手动。
> 
> vehicle.mode = VehicleMode（“ MANUAL”）

There is no MODE named "MANUAL" ,how to solve this issue?
Thanks a lot.