dronekit/dronekit-python #770: Can't change vehicle.mode

luuckiest, 2017-11-16:
Hi there:

    In the simulation, it is easy to change mode with a single line of code: VehicleMode("LAND"). It works perfectly fine, but when I connect a drone through the MAVLink using 
"mavproxy.exe --master="com5" --baudrate 57600 --out 127.0.0.1:14550" 
and then try to run the python code to change the mode, it only changes the mode the first time I use the command VehicleMode("GUIDED"), when I want to switch to VehicleMode("LAND"), it won't react.

AnkurSatya, 2017-11-27:
I am facing the same problem here. The drone reaches the target altitude and then does not react to the vehicle mode change command to land and falls down.
@luuckiest  did you have any luck?

luuckiest, 2017-11-27:
@AnkurSatya Okay, I was running the thing in windows and it did get me any further. And I have reinstalled every package: ). But!!!! I switch to Linux, with the same code, it just works. I don't know why but there is probably something wrong with the packages in windows though.