dronekit/dronekit-python #772: Dronekit goto convenience function - avoid rotation

JoseOcasio, 2017-11-21:
Hello, I have noticed that when using the goto convenience function for moving in meters, the drone faces the direction it is heading. Is it possible to move without rotating?

peterbarker, 2017-11-21:
On Tue, 21 Nov 2017, Jose Ocasio wrote:

> Hello, I have noticed that when using the goto convenience function for
> moving in meters, the drone faces the direction it is heading. Is it
> possible to move without rotating?

Not as-is,  If you send through a mav-cmd-comdition-yaw mavlink message 
first it should honour that.

https://github.com/ArduPilot/mavlink/blob/master/message_definitions/v1.0/common.xml#L861

JoseOcasio, 2017-11-21:
I do have a condition_yaw function, but I thought it gets overwritten by the goto.

```
def condition_yaw(heading, relative=False):
    """
    Send MAV_CMD_CONDITION_YAW message to point vehicle at a specified heading (in degrees).

    This method sets an absolute heading by default, but you can set the `relative` parameter
    to `True` to set yaw relative to the current yaw heading.

    By default the yaw of the vehicle will follow the direction of travel. After setting 
    the yaw using this function there is no way to return to the default yaw "follow direction 
    of travel" behaviour (https://github.com/diydrones/ardupilot/issues/2427)

    For more information see: 
    http://copter.ardupilot.com/wiki/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_condition_yaw
    """
    if relative:
        is_relative = 1 #yaw relative to direction of travel
    else:
        is_relative = 0 #yaw is an absolute angle
    # create the CONDITION_YAW command using command_long_encode()
    msg = vehicle.message_factory.command_long_encode(
        0, 0,    # target system, target component
        mavutil.mavlink.MAV_CMD_CONDITION_YAW, #command
        0, #confirmation
        heading,    # param 1, yaw in degrees
        0,          # param 2, yaw speed deg/s
        1,          # param 3, direction -1 ccw, 1 cw
        is_relative, # param 4, relative offset 1, absolute angle 0
        0, 0, 0)    # param 5 ~ 7 not used
    # send command to vehicle
    vehicle.send_mavlink(msg)
```

peterbarker, 2017-11-22:
On Tue, 21 Nov 2017, Jose Ocasio wrote:

> 
> I do have a condition_yaw function, but I thought it gets overwritten by the
> goto.

The condition yaw shouldn't be reset if you're already in guided mode.  Is 
it?

JoseOcasio, 2017-11-30:
Hello. I got a chance to test this. the goto function overwrites the yaw rotation command. 
So, the drone faces depending on where it is going. 

For example
goto (2, 0) will make the drone go north and face the drone north
goto (-2, 0) will make the drone go south and face the drone south

I also noticesd that the condition_yaw function sometimes works, sometimes it does not.

googl1, 2018-05-30:
If you are using Ardupilot, there is a parameter for that...
WP_YAW_BEHAVIOUR