dronekit/dronekit-python #776: vehicle.location.local_frame return none after arming [dronekit]

jessimaux, 2017-12-17:
im new in dronekit. My target is positioning drone in his local frame without global (gps). I arming drone in GUIDED_NOGPS mode and takeoff him. I make request for my position in local frame with next command: vehicle.location.local_frame, but it return 'north=none, west=none, down=none' What did i do wrong?

jessimaux, 2018-03-04:
It gets from local position estimator(accel and gyro)
But i can get it only in guided mode,no in guided_nogps
But i dont takeoff without gps in guided mode
For example,px4 firmware have offboard mode with lpe

pietrodn, 2018-03-17:
Maybe you're trying to read it too soon. Try passing `wait_ready=['local_frame']` to the [`connect`](http://python.dronekit.io/automodule.html#dronekit.connect) call.

canberkdurmus, 2018-11-28:
We are having the same issue, tried to call "connect()" `with wait_ready=['local_frame']` but api gives an exception `dronekit.APIException: wait_ready experienced a timeout after 30 seconds.` 
That means local frame never gets ready when vehicle not armed.
Shouldn't we try to get local_frame data when vehicle is armed?

pietrodn, 2018-11-28:
No. In PX4 the position estimate is independent of the armed/disarmed state. You shouldn't even arm the drone without a valid position estimate (that's quite unsafe!). I think your problem is in the PX4 configuration. Try to see if the drone gets a valid local position reference in QGroundControl.

canberkdurmus, 2018-11-28:
Sorry I forgot to point out that we are using APM 2.6 but I think the protocol is the same. Also we are getting the same exception on SITL.